RPi-Distro/chromium #2: --screenshot broken

Aarda, 2019-07-28:
chromium-browser  --window-size=1280,1024 --screenshot --headless https://raspberrypi.org

No screenshot is created.

RPI4 
chromium-browser --version
Chromium 74.0.3729.157 Built on Raspbian , running on Raspbian 10

spl237, 2019-07-28:
Is there some reason you are using chromium to get a screenshot rather than scrot, the tool included in Raspbian for that purpose?

XECDesign, 2019-07-28:
@spl237 It allows the user to take a screen shot in headless mode. For example, to render a web page into a png without actually running a desktop. I don't know what the actual use case is, but this has come up before and seems to be something some users want to do. I had a quick look, but couldn't find an obvious reason it wasn't working. The plan is to dig further when time allows.

Aarda, 2019-07-28:
@spl237. Yeah, chromium can run on a headless install (without X), e.g. Ubuntu server or Raspbian Lite.

Use case: 
- automate testing for web development, 
- create pictures for documentation.

BTW also creating screenshots with selenium is not working, probably related.

XECDesign, 2019-12-09:
New version which fixes this issue is available for testing here - https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=258892

closed 2020-05-28 by XECDesign