RPi-Distro/chromium #3: geo location not working

SG-aniki, 2019-09-29:
https://www.where-am-i.net/

if you visit this site in Rpi, it will not get the location,
however this site work on the same subnet on a chrome browser. any ideas?

XECDesign, 2019-09-30:
Not sure. Just in case, try disabling uBlock. But I suspect the problem may be on our end and related to the api keys we use.

SG-aniki, 2019-09-30:
@XECDesign thank you, will give it a go when I get back

XECDesign, 2019-09-30:
I've just tried ubuntu's keys and it worked, but we can't really use those for our distribution.

There are instructions on how to override keys here - https://www.chromium.org/developers/how-tos/api-keys

SG-aniki, 2019-09-30:
@XECDesign are we talking about the same thing? Visiting that page requires browser to share location, does it needs API?

XECDesign, 2019-09-30:
It needs the API since it use's google's location services

Related: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=934862

XECDesign, 2019-09-30:
After 10 minutes of clicking trying to get our own keys, I finally got to the point where it tells you geolocation requests cost money.

$5 per 1k requests or $4 if it's over 100k requests per month.

Very generous of Canonical to include that in their builds.

SG-aniki, 2019-09-30:
> After 10 minutes of clicking trying to get our own keys, I finally got to the point where it tells you geolocation requests cost money.
> 
> $5 per 1k requests or $4 if it's over 100k requests per month.
> 
> Very generous of Canonical to include that in their builds and explains why they ask other distributions not to use them.

you are fast! Kudos!

XECDesign, 2019-09-30:
Unfortunately, I think that means the answer is that geolocation is just not going to work in our builds. I have no idea how many requests we'd need to pay for or who else might use the keys.

SG-aniki, 2019-09-30:
@XECDesign  as long as we are able to input our own keys, it should be fine?

XECDesign, 2019-09-30:
Yeah, but that just means somebody else is paying for it. Or maybe Canonical have a deal with Google. Either way, I wouldn't publicise instructions on how to do it and what keys to use for that reason.

SG-aniki, 2019-09-30:
@XECDesign  I cant find the path in a raspberry pi 4 setup , i am talking about the browser?

On Chromium OS to specify the keys as environment variables append them to the end of /etc/chrome_dev.conf:
GOOGLE_API_KEY=your_api_key
GOOGLE_DEFAULT_CLIENT_ID=your_client_id
GOOGLE_DEFAULT_CLIENT_SECRET=your_client_secret

XECDesign, 2019-09-30:
I think I have to close this now. The answer is that we don't support geolocation right now.

closed 2019-09-30 by XECDesign

SG-aniki, 2019-10-07:
@XECDesign in raspberry pi 4 there is no such location and file?

/etc/chrome_dev.conf

SG-aniki, 2019-10-14:
@XECDesign any ideas

XECDesign, 2019-10-14:
AFAIK `/etc/chrome_dev.conf` is a Chrome/ChromiumOS thing.

Like I said, we don't support geolocation and I'm not comfortable providing instructions for stealing keys from somebody who pays for the service.

SG-aniki, 2019-10-14:
@XECDesign who tells you about stealing keys? I needed my chromium browser to have the geolocation to work with selium to grab the current location

SG-aniki, 2019-10-14:
@XECDesign i got a google dev account with my own api key, i need the location for chromium browser to work correctly for my python script to grab current location

XECDesign, 2019-10-14:
Chromium's own documentation covers this in detail.