RPi-Distro/chromium #4: Browser version 74 while the latest stable version is 76

yosolan, 2019-09-29:
Unable to run Netflix on Chromium due to the version issue in RPI - Chromium. Its working fine in Chromebook. Please refer to 
https://blog.vpetkov.net/2019/07/12/netflix-and-spotify-on-a-raspberry-pi-4-with-latest-default-chromium/

XECDesign, 2019-09-29:
Sometimes it works, sometimes it doesn't, but we don't have a license to distribute widevine, so we can't offer any support for Netflix.

If that blog is correct, then the issue should resolve itself the next time we update Chromium.

yosolan, 2019-09-29:
Yes, understood, thanks for the prompt response. 
When would the update happen in chromium on RPi?.

XECDesign, 2019-09-29:
Sorry, I don't have an answer for that. I don't think it's soon.

XECDesign, 2019-12-09:
New version available for testing here - https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=258892

closed 2020-05-28 by XECDesign