RPi-Distro/chromium #5: Problems with XCB/X11 when building

EwanRoycroft, 2019-12-11:
I'm trying to build Chromium v74 from source, implementing the MMAL patch, and received the following error when building:
```
ninja: Entering directory `out/Release_GN_arm'
[1/27145] SOLINK ./libGLESv2.so
FAILED: libGLESv2.so libGLESv2.so.TOC 
python "../../build/toolchain/gcc_solink_wrapper.py" --readelf="arm-linux-gnueabihf-readelf" --nm="arm-linux-gnueabihf-nm" --sofile="./libGLESv2.so" --tocfile="./libGLESv2.so.TOC" --output="./libGLESv2.so" -- ../../third_party/llvm-build/Release+Asserts/bin/clang++ -shared -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=lld -Wl,--icf=all -Wl,--color-diagnostics --target=arm-linux-gnueabihf -Werror -Wl,-O2 -Wl,--gc-sections -rdynamic -nostdlib++ --sysroot=../../build/linux/debian_sid_arm-sysroot -L../../build/linux/debian_sid_arm-sysroot/usr/local/lib/arm-linux-gnueabihf -L../../build/linux/debian_sid_arm-sysroot/lib/arm-linux-gnueabihf -L../../build/linux/debian_sid_arm-sysroot/usr/lib/arm-linux-gnueabihf -o "./libGLESv2.so" -Wl,-soname="libGLESv2.so" @"./libGLESv2.so.rsp"
ld.lld: error: unable to find library -lxcb-image
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
I copied `libxcb-image.so` from the RPi4 and placed it in `./build/linux/debian_sid_arm-sysroot/lib/arm-linux-gnueabihf/`; which seemed to work, and ninja carried on compiling.

I then received the following errors relating to one of the files created by `v74.0.3729.182_mmal_3_22.patch`:
```
ninja: Entering directory `out/Release_GN_arm'
[36/14204] CXX obj/components/viz/service/service/service_jumbo_2.o
FAILED: obj/components/viz/service/service/service_jumbo_2.o ../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/components/viz/service/service/service_jumbo_2.o.d -DVIZ_SERVICE_IMPLEMENTATION -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DNO_TCMALLOC -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"354873-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -DCR_LIBCXX_REVISION=355550 -DCR_LIBCXXABI_REVISION=354284 -D_LIBCPP_ENABLE_NODISCARD -DCR_SYSROOT_HASH=ef5c4f84bcafb7a3796d36bb1db7826317dde51c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_32 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_26 -DWEBP_EXTERN=extern -DGL_GLEXT_PROTOTYPES -DUSE_GLX -DUSE_EGL -DSK_HAS_PNG_LIBRARY -DSK_HAS_WEBP_LIBRARY -DSK_HAS_JPEG_LIBRARY -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DUCHAR_TYPE=uint16_t -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DHAVE_PTHREAD -DLEVELDB_PLATFORM_CHROMIUM=1 -DLEVELDB_PLATFORM_CHROMIUM=1 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_CHROMIUM_BUILD -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DNO_MAIN_THREAD_WRAPPING -I. -I../.. -Igen -I../../third_party/libwebp/src -I../../third_party/khronos -I../../gpu -I../../third_party/libyuv/include -I../../skia/config -I../../skia/ext -I../../third_party/skia/include/c -I../../third_party/skia/include/codec -I../../third_party/skia/include/config -I../../third_party/skia/include/core -I../../third_party/skia/include/docs -I../../third_party/skia/include/effects -I../../third_party/skia/include/encode -I../../third_party/skia/include/gpu -I../../third_party/skia/include/pathops -I../../third_party/skia/include/ports -I../../third_party/skia/include/utils -I../../third_party/skia/src/gpu -I../../third_party/skia/src/sksl -I../../third_party/skia/modules/skottie/include -I../../third_party/ced/src -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -I../../third_party/protobuf/src -I../../third_party/mesa_headers -I../../third_party/libwebm/source -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/leveldatabase -I../../third_party/leveldatabase/src -I../../third_party/leveldatabase/src/include -I../../third_party/boringssl/src/include -I../../third_party/webrtc_overrides -I../../third_party/webrtc -Igen/third_party/webrtc -I../../third_party/abseil-cpp -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -fcomplete-member-pointers --target=arm-linux-gnueabihf -march=armv7-a -mfloat-abi=hard -mtune=generic-armv7-a -no-canonical-prefixes -mfpu=neon -mthumb -Wall -Werror -Wextra -Wimplicit-fallthrough -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/glib-2.0 -isystem../../build/linux/debian_sid_arm-sysroot/usr/lib/arm-linux-gnueabihf/glib-2.0/include -Wno-header-guard -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/nss -isystem../../build/linux/debian_sid_arm-sysroot/usr/include/nspr -std=c++14 -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../build/linux/debian_sid_arm-sysroot -fvisibility-inlines-hidden -c gen/components/viz/service/service_jumbo_2.cc -o obj/components/viz/service/service/service_jumbo_2.o
In file included from gen/components/viz/service/service_jumbo_2.cc:42:
./../../components/viz/service/display_embedder/shared_image_x11.cc:226:17: error: out-of-line definition of 'check_support' does not match any declaration in 'viz::SharedImageX11'
SharedImageX11::check_support(Display *const display)
                ^~~~~~~~~~~~~
./../../components/viz/service/display_embedder/shared_image_x11.cc:231:35: error: no matching function for call to 'XGetXCBConnection'
  xcb_connection_t * const xcbc = XGetXCBConnection(display);
                                  ^~~~~~~~~~~~~~~~~
../../build/linux/debian_sid_arm-sysroot/usr/include/X11/Xlib-xcb.h:13:19: note: candidate function not viable: no known conversion from 'viz::Display *const' to 'Display *' (aka '_XDisplay *') for 1st argument
xcb_connection_t *XGetXCBConnection(Display *dpy);
                  ^
2 errors generated.
[45/14204] CXX obj/content/browser/browser/browser_jumbo_5.o
ninja: build stopped: subcommand failed.
```

What might be causing these errors? Am I missing more dependencies from the sysroot?

XECDesign, 2019-12-11:
This repo is for the packaging only, I can't provide general build support. All I can say is that I was able to get it to cross-compile using the instructions included in the patch a long time ago, before I switched to native builds.

EwanRoycroft, 2019-12-11:
How long does native compiling take? Must be pretty intensive given the recommended system requirements for building Chromium.

I opened a [topic on the forums](https://www.raspberrypi.org/forums/viewtopic.php?t=258998) the other day looking for up-to-date instructions for building/patching Chromium. Do you know of any? Is this something that might be released with the new v78?

XECDesign, 2019-12-11:
It took 2 days on an 8 core 2.4GHz ARM server. You'll run out of RAM trying to do it on a pi unless you use the 64bit kernel and some swap. Needless to say it will take even longer.

closed 2019-12-12 by XECDesign