RPi-Distro/chromium #6: MMAL incompatibility with custom chromium builds

RomanValov, 2020-04-26:
Hello,

I'm developing application using Qt framework and it's QtWebEngine component which in turn is based on Chromium browser. I've prepared custom Qt 5.14 build to run on my RPi4 (with Raspbian Buster).

I know that FFmpeg bundled with Buster is patched to support hardware accelerated H.264 playback using MMAL interface. And there is an option for Qt build to use system-wide FFmpeg library for QtWebEngine (not one bundled inline and used by default).

However even with the option used I found H.264 videos playback to consume a lot of CPU and being not so fluent as with Raspbian-bundled Chromium browser. Finally it seems the reason is that Raspbian-bundled Chromium browser is also additionally patched to support playback with MMAL.

Initially I've considered to apply the MMAL patch to Qt codebase but Chromium versions differ. Patch doesn't apply cleanly and I consider porting and maintaining it to be unreasonably difficult to me.

I wonder if there is a chance to make usage of FFmpeg MMAL playback more transparent for Chromium, so no additional patches will be required for that?

Is there are any other ways to provide FFmpeg MMAL playback for custom Chromium builds?

XECDesign, 2020-05-28:
I think they may be some work being done in that area as we move away from interfaces like MMAL. This is probably a bit outside the scope of this issue tracker, but @popcornmix may have a comment on this.

viraniac, 2020-06-14:
Does that mean MMAL is being deprecated? Is there a place where we can read about how its getting replaced as that might help the developers to update their applications accordingly?

popcornmix, 2020-06-15:
DRM for display and V4L2 for video codecs is the preferred route for the future.
These are standard linux interfaces, so generic code should be possible that runs on a range of linux architectures.

closed 2022-01-19 by XECDesign