RPi-Distro/chromium #7: nacl?

Botspot, 2020-05-28:
It appears the NaCl API is disabled. Why?
Without it, common Chrome Apps like Zoom and VNC don't work.
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=274409

XECDesign, 2020-05-28:
@jc-kynesim Any idea?

XECDesign, 2020-05-28:
I believe we inherit that flag from debian and ubuntu

https://salsa.debian.org/chromium-team/chromium/-/blob/master/debian/rules#L67

Not sure why they disable it though.

XECDesign, 2020-05-28:
Looks like it's disabled because it's deprecated in favour of WebAssembly:

https://developer.chrome.com/native-client/migration

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=655536

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=943346

https://bugs.launchpad.net/ubuntu/+source/chromium-browser/+bug/882942

Botspot, 2020-05-28:
I'd like to try compiling chromium-browser with nacl enabled.
Is there a good tutorial out there on how to compile chromium-browser with the RPi optimizations?

XECDesign, 2020-05-28:
The README in the patch gives some hints on how to cross-compile it.

I wouldn't wish compiling chromium natively on my worst enemy, so I'll kick off a build with the flag changes and see what happens.

XECDesign, 2020-05-28:
```
ERROR at //build/config/ui.gni:50:1: Assertion failed.
assert(!use_glib || (is_linux && !is_chromeos && !is_chromecast))
^-----
See //build/config/BUILD.gn:21:1: whence it was imported.
import("//build/config/ui.gni")
^-----------------------------
See //build/config/BUILDCONFIG.gn:580:19: which caused the file to be included.
        deps += [ "//build/config:${_target_type}_deps" ]
                  ^------------------------------------
Traceback (most recent call last):
  File "tools/gn/bootstrap/bootstrap.py", line 135, in <module>
    sys.exit(main(sys.argv[1:]))
  File "tools/gn/bootstrap/bootstrap.py", line 130, in main
    '--args=%s' % gn_gen_args, "--root=" + SRC_ROOT
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/<<PKGBUILDDIR>>/out/Release/gn', 'gen', '/<<PKGBUILDDIR>>/out/Release', '--args= enable_hangout_services_extension=true enable_mdns=true enable_nacl=true enable_wayland_server=false enable_widevine=true fieldtrial_testing_like_official_build=true is_component_build=false is_component_ffmpeg=true is_debug=false is_desktop_linux=true is_official_build=true blink_symbol_level=0 symbol_level=1 treat_warnings_as_errors=false use_allocator="none" use_alsa=true use_aura=true use_cups=true use_dbus=true use_gio=true use_glib=true use_gold=false use_libpci=true use_pulseaudio=true use_gnome_keyring=false use_sysroot=false use_system_harfbuzz=false use_system_libjpeg=false rtc_enable_protobuf=false rtc_use_h264=true is_clang=true clang_base_path="/usr" clang_use_chrome_plugins=false use_custom_libcxx=false use_lld=false is_cfi=false use_thin_lto=false fatal_linker_warnings=false target_os="linux" current_os="linux" optimize_webui=false cc_wrapper="ccache" target_cpu="arm" arm_float_abi="hard" arm_optionally_use_neon=false google_api_key="AIzaSyCkfPOPZXDKNn8hhgu3JrA62wIgC93d44k" google_default_client_id="811574891467.apps.googleusercontent.com" google_default_client_secret="kdloedMFGdGla2P1zacGjAQh" is_debug=false', '--root=/<<PKGBUILDDIR>>']' returned non-zero exit status 1
```

trejan, 2020-05-28:
Not sure why they made the [assertion](https://chromium.googlesource.com/chromium/src/+/master/build/config/ui.gni#47) fail if glibc isn't enabled. The comment for it implies it should only fail if glibc is enabled for ChromeOS or Chromecast. glibc being disabled for this build is probably related to [is_nacl_glibc=false](https://github.com/RPi-Distro/chromium-browser/blob/master/debian/known_gn_gen_args-armhf#L119). This is looking like a dead end as [glibc dropped NaCl support in 2017](https://sourceware.org/git/?p=glibc.git;a=commit;h=42a844c6a213f9219a4baa013c7305679d5dcaaa).

jc-kynesim, 2020-05-29:
The README - whilst not always up to date - is normally quite close to accurate.  I cut and paste from it as part of my normal development cycle so it is in my interest for it to be good.  It is likely to be weakest (read wrong) on its initial setup instructions where the fact that I always have an existing tree means that I hardly ever use that bit.
If you spot stuff in it that is obviously wrong I'll fix it for future reference.

XECDesign, 2020-05-29:
For reference - https://github.com/RPi-Distro/chromium-browser/blob/master/debian/patches/v78.0.3904.126_mmal_3_23.patch#L8627

XECDesign, 2020-05-29:
But I think in general this is a wontfix, since it seems to be a deprecated feature.

Botspot, 2020-05-29:
> since it seems to be a deprecated feature.

Native Client is supported in x86 versions of chromium. And in Chrome.
The VNC and Zoom chrome apps work in them.

Native Client is only deprecated from _websites_ running it.

XECDesign, 2020-05-29:
It's not enabled on any version of chromium in the major distros and we're just following them. Trying to enable it doesn't work. If somebody else wants to maintain a patch that makes it work we can consider pulling it in.

closed 2020-05-29 by XECDesign

Botspot, 2020-05-30:
I'm pretty sure _it is_ enabled in other distros. For x86, anyway. Otherwise, how could those chrome extensions be working on them? (Zoom and VNC chrome apps work flawlessly in an x86 debian laptop, but not on a Pi)  

Do you mean "_all armhf distros_ don't have nacl"?

XECDesign, 2020-11-03:
Just had another go at this and still no luck. Even when I have a build with NaCl enabled, it errors out with "nacl process exited with status 5"

Botspot, 2020-11-03:
> Just had another go at this and still no luck. Even when I have a build with NaCl enabled, it errors out with "nacl process exited with status 5"

That's OK, there's other ways to run Zoom now. One is [through accelerated emulation](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=287680) with the x86 Linux app, and the other uses ChromiumOS (FydeOS) which recently added NaCl upon my request.  
The emulated version is surprisingly smooth, and works better than any other option.  
If you want to continue trying to get NaCl support, I suggest you talk to the makers of FydeOS, who succeeded in getting it to work.

XECDesign, 2020-11-03:
Is using Zoom directly through the website rather than through the WebStore worse? Why is the app something people are after?

Botspot, 2020-11-03:
Yes. It's missing features, and I think the NaCl code makes it run smoother.

But again, either of those pale in comparison to emulating the linux app, which is much faster, and has Gallery View, virtual backgrounds, etc.

XECDesign, 2020-11-03:
Got it, thanks.

Botspot, 2020-11-03:
On the other hand, there are some things that the Zoom Chrome App can do, that the emulated Zoom cannot.

Like signing in with Google, for example. See https://github.com/Botspot/pi-apps/issues/57

Also, if Raspbian's Chromium had NaCl, other useful Chrome apps would work, including Mindstorms and VNC Viewer.

So I would be of the opinion that NaCl would still be a good thing, even in light of other Zoom alternatives. Does that make sense?

XECDesign, 2020-11-03:
Yeah, I'm not against NaCl, I am just not sure how much effort to put into something that doesn't just work and may end up being a week-long rabbit-hole when there are many other things to do.

Botspot, 2020-11-03:
> Yeah, I'm not against NaCl, I am just not sure how much effort to put into something that doesn't just work and may end up being a week-long rabbit-hole when there are many other things to do.

Completely understand.  
Again, I highly suggest you go ask how the FydeOS devs did it. Maybe there's a file that needs some lines changed and then it'll work? See https://github.com/FydeOS/chromium_os-raspberry_pi/issues/72