RPi-Distro/chromium #8: Out of Date

g1gsw, 2020-11-13:
When I try to install the latest gmail notifier it tells me that chromium is out of date and does not work.  If I leave it open on the desktop for a while google keeps informing me that it is out of date and a lot of things do not work becuse os this.

XECDesign, 2020-11-16:
There seems to be a number of things that call themselves gmail notifier. Do you have a link to what you're using and how you installed it?

g1gsw, 2020-11-16:
Hi here is the link.

https://chrome.google.com/webstore/detail/checker-plus-for-gmail/oeopbcgkkoapgobdbedcemjljbihmemj?utm_source=chrome-ntp-icon

Colin

XECDesign, 2020-11-16:
I don't see any information there to see how it determines what's too old. I found a page where he says he only supports the 'stable' version of chromium, but not much else. It doesn't look like anything we can fix on our end, unfortunately.

closed 2020-12-07 by XECDesign