RPi-Distro/chromium #9: Up-to-date RPI OS chromium browser patches to download?

systemq2, 2020-11-21:
Hi there,

I'd like to use the CEF library with mmal patches for the chromium browser applied on RPI 3b+ for my project. Unfortunately the latest patch version published in this repository is 78.0.3904.108 whereas the chromium browser version in the current RPI OS is 84.0.4147.141. Would it be possible to publish up-to-date RPI patches for chromium browser to give other people a chance to build a mmal-enabled libcef and use the hardware decoding capability of the RPI hardware?

XECDesign, 2020-11-23:
Added https://github.com/RPi-Distro/chromium-browser/blob/master/debian/patches/v84.0.4147.141_mmal_3.31.patch and the other patches used for that build.

systemq2, 2020-11-24:
Thank you, XECDesign. I appreciate that and your contribution to the RPI project in general.

XECDesign, 2020-11-24:
Thank you. The real work here was done by jc-kynesim, the firmware and kernel devs, but I'll gladly take the credit.

closed 2020-11-24 by XECDesign