RPi-Distro/chromium #10: CSS filters on divs are rendering black

audas, 2020-11-26:
This is failing with the latest kernel 5.10 and native gl renderer 

dtoverlay=vc4-kms-v3d.

This was an issue in Chrome several years ago with Nividia - removing chrome://flags use GPU hardware rendering removed the problem.

XECDesign, 2020-11-26:
@naushir One for Igalia?

popcornmix, 2020-11-27:
@audas can you provide a url that exhibits the issue?

audas, 2020-11-27:
https://display-screen.media/display.php?id_presentations_manifests=16

The images at the top and "education funding" third from bottom - these are being animated VERY SLOWLY to give a chance to inspect them. They are all the same text etc. Top images should be transparent background.

The reason there is a difference between other texts is the filter and these are all divs within divs. But again - when I switched off hardware acceleration in flags in chrome it cleared this up. (I need hardware acceleration though).

I just changed the chromium-browser back to 78 and it fixed the issue.

XECDesign, 2022-01-19:
Is this still an issue in Bullseye chromium-browser? That page seems to render the same way there as it does in firefox on my desktop.

closed 2022-02-24 by XECDesign