RPi-Distro/chromium #12: Occasional complete crash from printing

mpratt14, 2020-12-31:
Raspberry Pi 4 Model B Rev 1.1
Linux 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
Chromium 86.0.4240.197 Built on Raspbian , running on Raspbian 10
hpcups 3.18.12

On several websites, but not all, printing the page crashes the entire program sometimes
for example tripadvisor.com but not google.com
happens more often when an ad-blocker is used,
for example when using Disconnect or Ghostery
but the crashes still happen sometimes when the page is unblocked

I have tried collecting logs but they are so long and nothing useful at the end

mpratt14, 2021-01-21:
It is also happening on Vivaldi Linux ARM version....

I will post to bugs.chromium...

XECDesign, 2022-01-19:
I'm not seeing this in v95 on Bullseye. Is this still an issue?

closed 2022-02-24 by XECDesign