RPi-Distro/chromium #13: Unable to share screen/windows while in google meet

jcochran26, 2021-01-22:
As described, when using 64-bit RPI OS, I'm unable to share my screen or any windows while using Chromium.   Note that this works fine in FireFox ESR, so I tend to think the issue is with the browser, not the OS.

Also worth noting, and maybe related, my webcam microphone stays muted by default, and attempts to unmute it fail (it just mutes itself again).   Video works fine, however.   And, as before, this also works fine with FireFox ESR.

XECDesign, 2021-01-22:
The 64 bit install doesn't use our build of chromium yet. This is straight from Raspbian/Debian.

Roburetto, 2021-02-11:
This is also happening in RPI OS 32 bits and is a serious problem.

chromium-browser/testing,now 86.0.4240.197-rpt1 armhf [installed]
  Chromium web browser, open-source version of Chrome

There's an old Debian issue regarding "enable_hangout_services_extension" flag being disabled in rules, not sure if it relates: 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=886358

XECDesign, 2021-02-11:
That flag should be enabled in the next release (for the armhf build).

XECDesign, 2021-03-16:
Should be fixed in v88.

closed 2021-03-16 by XECDesign