RPi-Distro/chromium #14: Hardware acceleration causes problems with PowerPoint Online Slideshow

wce-pi, 2021-01-26:
When hardware acceleration switch is enabled, text in PowerPoint slideshows (via Office 365 online) is rendered incorrectly.  This has been tested and seen with both 32-bit and 64-bit ARM distributions of Raspberry Pi OS on RPi 4 and RPi 400.  The issue does not exist in the x86 distribution.  Images are attached.  Let me know if there's anything I can test to help resolve.

PPT in Chromium - designer mode:
[image]

PPT in Chromium - slideshow:
[image]

Nabla128k, 2021-04-12:
Confirming this - it also affects Teams shared presentation - if remote side on windows shares a presentation thru teams, its displayed garbled as this

XECDesign, 2022-01-19:
Is this still an issue with chromium-browser v95 in Bullseye?

closed 2022-02-24 by XECDesign