RPi-Distro/chromium #16: --enable-gpu-memory-buffer-compositor-resources can't be used when X11 server is run without sudo privileges

korbav, 2021-03-20:
I just noticed that as soon as the `--enable-gpu-memory-buffer-compositor-resources` switch is enabled to launch chromium-browser within an X11 session opened without the sudo privileges, the browser won't render anything except fuzzy black squares on a white background. (As soon as the switch is removed, everything works).

I've been able to detect some errors messages that led me to think that this has to be related to the GPU not being used at all in that case.

```
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.547594:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.550619:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.745367:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.752825:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.760321:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.767786:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.769448:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: usrvcsm: [vcsm_malloc_cache]: [3773]: mmap FAILED (hdl: 0)
Mar 18 17:11:36 raspberrypi sh[3672]: usrvcsm: [vcsm_free]: [3773]: invalid device or handle!
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0100/000000.946805:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(276)] Failed to create pixmap 320x320, RGBA_8888, usage SCANOUT
```

And 

```
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.533209:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.700064:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglCreateImageKHR: _eglParseImageAttribList
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.700414:ERROR:gl_image_egl.cc(37)] Error creating EGLImage: EGL_BAD_PARAMETER
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.700737:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(283)] Failed to create GLImage 320x320, RGBA_8888, usage SCANOUT
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.701820:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglCreateImageKHR: _eglParseImageAttribList
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.702198:ERROR:gl_image_egl.cc(37)] Error creating EGLImage: EGL_BAD_PARAMETER
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.702654:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(283)] Failed to create GLImage 320x320, RGBA_8888, usage GPU_READ
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.703055:ERROR:shared_image_backing_gl_texture.cc(535)] CreateSharedImage: Failed to create bindable image
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.703293:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.703524:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.711244:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.749787:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:35 raspberrypi sh[3672]: [3773:3773:0318/171135.753224:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
...
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.388059:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglCreateImageKHR: _eglParseImageAttribList
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.388421:ERROR:gl_image_egl.cc(37)] Error creating EGLImage: EGL_BAD_PARAMETER
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.388742:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(283)] Failed to create GLImage 320x320, RGBA_8888, usage SCANOUT
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.390238:ERROR:gl_surface_egl.cc(773)] EGL Driver message (Error) eglCreateImageKHR: _eglParseImageAttribList
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.390595:ERROR:gl_image_egl.cc(37)] Error creating EGLImage: EGL_BAD_PARAMETER
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.390979:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(283)] Failed to create GLImage 320x320, RGBA_8888, usage GPU_READ
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.391410:ERROR:shared_image_backing_gl_texture.cc(535)] CreateSharedImage: Failed to create bindable image
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.391674:ERROR:shared_image_factory.cc(553)] CreateSharedImage: could not create backing.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.391952:ERROR:shared_image_stub.cc(201)] SharedImageStub: Unable to create shared image
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.398271:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
...
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.436893:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.440393:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.524268:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0318/171136.527819:ERROR:shared_image_manager.cc(214)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
Mar 18 17:11:36 raspberrypi sh[3672]: usrvcsm: [vcsm_malloc_cache]: [3773]: mmap FAILED (hdl: 0)
Mar 18 17:11:36 raspberrypi sh[3672]: usrvcsm: [vcsm_free]: [3773]: invalid device or handle!
Mar 18 17:11:36 raspberrypi sh[3672]: [3773:3773:0100/000000.946805:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(276)] Failed to create pixmap 320x320, RGBA_8888, usage SCANOUT
```

XECDesign, 2022-01-19:
@jc-kynesim I'm seeing this as well. Is `--enable-gpu-memory-buffer-compositor-resources` something that makes sense to use and should be supported?

jc-kynesim, 2022-01-19:
As a first reaction: "not a clue". Chrome has more switches than it is possible to list never mind test - in general you get switches enabled by default that google actually supports for the platform you are on.  I'll grant that that switch looks superficially attractive but I'm guessing it has hit a wall of "not implemented" somewhere.
I'll have a quick look for the next version as at least some of what is appears to be failing there should probably work, google seems to  still be doing a certain amount of catchup in their switch to ozone.

XECDesign, 2022-01-19:
I tried the same with the official v97 snap and haven't had any luck there either. I'll leave it open for now, but it sounds like something we won't end up supporting.

jc-kynesim, 2022-01-24:
In my dev v98 build on current bullseye either that flag works or it does nothing - it doesn't produce errors. But I've still got some things to look at in terms of GL buffer allocation.

XECDesign, 2024-10-16:
Is this still an issue? Just tried it under wayland on bookworm and I'm not seeing any error spam (other than the usual).

closed 2024-10-18 by XECDesign