RPi-Distro/chromium #19: Chromium 88 very strange copy & paste behaviour

Pearl-852, 2021-05-25:
There is this issue being discussed in the [Raspberry Pi Forum](https://www.raspberrypi.org/forums/viewtopic.php?p=1858297) 

I am experiencing the same problem also, characters that are utf-8 encoded fonts are displaying as a /u+nnnn code when pasted into Chromium browser (i.e. Google Translate, Google search, etc.)

**What I have tested so far,**
1. Copy a Chinese filename from PCManFM, and paste into Google Translate, shows /u+nnnn codes.
2. Copy some Chinese characters from a email in Thunderbird, and paste into Google Translate, shows /u+nnnn codes.
3. Copy some Chinese characters from a document in LibreOffice Writer, and paste into Google Translate, shows just "????"
4. Copy some Chinese characters from a webpage in Chromium browser, and paste into Google Translate, shows correctly.

**My current workaround is**
1. Copy the desire unicode characters as usual.
2. Paste the copied unicode contents into a new document to Google Doc in Chromium browser.
3. Highlight and copy the same pasted contents from Google Doc.
4. Paste this newly copied contents into Google Translate, Google search, etc. and it should display correctly.

XECDesign, 2021-05-25:
Not sure if this is a generic issue with chromium 88 that will go away once we update to 90. We'll have to see when v90 is ready.

XECDesign, 2021-05-25:
Found a workaround for you.

Go to `chrome://flags/#raw-clipboard`, enable it and relaunch chromium. That seems to make it work as you expect.

Pearl-852, 2021-05-25:
Thanks for your quick response. Tried with your workaround suggestion, but the results are the same as before. See screenshots below,

**Flag enabled:**
[image]

**Test results:** (Same orders as reported on my original post)
[image]

XECDesign, 2021-05-25:
Did you relaunch chromium after making that change? Because it's working for me.

Pearl-852, 2021-05-25:
Yes, I did. Twice.

XECDesign, 2021-05-25:
Wonder why we're seeing different behaviour. Sounds like we'll need to see what happens with v90.

Pearl-852, 2021-05-25:
Even, rebooted system just now to make sure.

Pearl-852, 2021-11-24:
For the record, the issue has been fixed on Chromium Browser 
**Version 90.0.4430.212 (Developer Build) built on Debian 10.2, running on Debian 10.11 (32-bit)**

closed 2021-11-24 by Pearl-852