RPi-Distro/chromium #20: Chromium-browser won't stay logged into google websites

PeriBelph, 2021-05-29:
I've made a fresh Raspberry Pi OS lite installation on my Raspberry Pi 3B+ rev. 1.2 three days ago.
I've ran `sudo apt-get update && sudo apt-get upgrade -y && sudo apt-get distr-upgrade -y` when first booting in so it should all be recent enough.

I'm trying to use chromium-browser for my daily browsing but for some reason only on google website I won't stay logged in after restarting the browser. Reddit, Twitter, Twitch, Discord all keep me logged in, so I think cookies work fine. I tested firefox to see if it works there indeed in firefox I just stay logged into YouTube and Gmail.
I would use firefox but it seems to perform way worse than chromium-browser on the Raspberry Pi 3B+. If there's a way to make it perform as good or better than chromium-browser I'd switch immediately.

Chromium-settings:
Cookies are allowed, even in incognito.
Browser data is not set to clear automatically after closing.
Auto Sign-in is enabled.
Offer to save passwords is enabled. But google websites don't even ask me if I want to save.

I already tried:
Resetting browser to default settings via Settings > Advanced > Reset settings.
Clear browsing data.
Deleting ~/.config/chromium and ~/.cache/chromium.
I even reinstalled the entire OS the first day and not even that worked.

OS: Raspberry Pi OS lite 32-bit (Raspbian GNU/Linux 10 (Buster) armv7)
Host: Raspberry Pi 3 Model B Rev 1.2
Kernel: 5.10.17-v7+
CPU: BCM2835 (4) @ 1.200GHz
Chromium Version: 88.0.4324.187 (Official Build) Built on Raspbian, running on Raspbian 10 (32-bit)
Package Name: chromium-browser/testing, now 88.0.4324.187-rpt1 armhf

Additional: Every time I start the browser it tells me that "Chromium didn't shut down correctly", I am not sure if this is related but thought it might be helpful.

I hope I provided all necessary information, I can provide anything that might be needed to fix this.

<3 Belph

Jai-JAP, 2021-06-23:
Itoo have exactly the same issue on RPi4. I have same settings as above and also the same version.

Also I get the same message "Chromium didn't shut down correctly"

This issue started after updating to chromium v88 before this this issue was not present, but the message "Chromium didn't shut down correctly" was always shown.

XECDesign, 2021-06-23:
Could you launch it through `chromium-browser` from a terminal window, log into the site you wanted to visit, close the chromium windows and then post the output?

Jai-JAP, 2021-07-04:
> Could you launch it through `chromium-browser` from a terminal window, log into the site you wanted to visit, close the chromium windows and then post the output?

```
pi@raspberrypi:~ $ chromium-browser
Gtk-Message: 13:52:27.459: Failed to load module "xapp-gtk3-module"
[1939:1995:0704/135229.142483:ERROR:nss_util.cc(284)] After loading Root Certs, loaded==false: NSS error code: -8018
[1989:1989:0704/135229.474772:ERROR:gl_surface_presentation_helper.cc(259)] GetVSyncParametersIfAvailable() failed for 1 times!
[1989:1989:0704/135316.373225:ERROR:gl_surface_presentation_helper.cc(259)] GetVSyncParametersIfAvailable() failed for 2 times!
Trace/breakpoint trap
 ```

Jai-JAP, 2021-09-09:
@XECDesign ^

XECDesign, 2021-10-01:
I've been trying to reproduce it, but no luck.

You could also try running it with `--enable-logging=stderr --v=2` which may provide more information. Is there anything in dmesg when chromium closes?

v92, has recently gone out. Is it exhibiting the same problem?

Jai-JAP, 2021-10-01:
I am currently away from my pi so I haven't tested chromium v92.

I will try this after I go home.

Jai-JAP, 2021-10-04:
Still similar results in v92

Except this time it flashes the signed in mail list for a while and then redirects to login page

Logs:
```
pi@raspberrypi:~ $ chromium-browser 
[21391:21391:1004/111834.185740:ERROR:component_loader.cc(186)] Failed to parse extension manifest.
[21443:21443:1004/111834.745862:ERROR:gpu_init.cc(441)] Passthrough is not supported, GL is egl
[21443:21443:1004/111835.493789:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[21443:21443:1004/111904.474151:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111904.478084:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111905.965910:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111905.970158:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111908.664328:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111908.666774:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111913.754742:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111913.943499:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111914.136953:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111922.498125:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111923.115076:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111923.490165:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111923.741769:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111924.082374:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111924.215436:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111924.794810:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111924.989772:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111925.664679:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111925.830127:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111925.953366:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111926.640327:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111926.790594:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111927.321594:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111927.441655:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111932.342805:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
[21443:21443:1004/111932.876772:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
[21443:21443:1004/111938.057262:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
[21443:21443:1004/112001.615742:ERROR:shared_image_representation.cc(207)] Attempt to read from an uninitialized SharedImage
```
Also v92 has random white patches as in below picture
[image]

The blur effect is not the issue as image is edited as it had some sensitive data
See the white patch at the centre

XECDesign, 2021-10-04:
@jc-kynesim Any idea what might be happening there? Insufficient CMA memory maybe?

jc-kynesim, 2021-10-04:
I'm going to guess that the "chromium didn't shut down correctly" message is the culprit. for the lack of persistent login.  The white square is something I haven't seen personally - I'd certainly try more CMA to see if it changed anything.  I'll try to see if I can recreate this.

XECDesign, 2021-10-04:
But why does it think it didn't shut down properly? Seems like it's crashing on exit rather than closing cleanly or something. But still, you'd think the sessions should survive even that scenario.

jc-kynesim, 2021-10-04:
I don't know & I agree with you.

jc-kynesim, 2021-10-04:
Still don't know why this is happening, I think I'd always assumed that relogin was required for security, but given that my id is cached on my windows m/c I guess it should be cached on my Pi.  I'll investigate when I have a bit of time.

Jai-JAP, 2021-10-04:
> I'm going to guess that the "chromium didn't shut down correctly" message is the culprit. for the lack of persistent login. The white square is something I haven't seen personally - I'd certainly try more CMA to see if it changed anything. I'll try to see if I can recreate this.

yeah it may be the culprit as the message still comes popping up randomly

Jai-JAP, 2021-10-04:
Also this time with v92 google sites get logged out after some time even if the chromium browser is running constantly
This doesnt happen with non-google sites

Jai-JAP, 2021-10-04:
> I've been trying to reproduce it, but no luck.
> 
> You could also try running it with `--enable-logging=stderr --v=2` which may provide more information. Is there anything in dmesg when chromium closes?
> 
> v92, has recently gone out. Is it exhibiting the same problem?

Sorry the above logs are without `--enable-logging=stderr --v=2` the below is the log with given flags
[chromium-log.txt](https://github.com/RPi-Distro/chromium-browser/files/7278431/chromium-log.txt)

XECDesign, 2021-10-04:
`sqlite_persistent_store_backend_base.cc(171)] Failed to post task from FlushAndNotifyInBackground@net/extras/sqlite/sqlite_persistent_store_backend_base.cc:220 to client_task_runner_.`

looks like it might be relevant.

slokhorst, 2021-12-09:
Problem still present in chromium-browser 95.0.4638.78-rpt1 on Raspian 11 (32-bit). I do *not* see any messages regarding "chromium didn't shut down correctly".

XECDesign, 2021-12-10:
I think this is related to the browser's sign in feature.

If you go to the "You and Google" page in the settings and disable "Allow Chromium sign-in" do logins persist?

slokhorst, 2021-12-11:
I think the issue has to do with misconfigured Google API keys. See [this Chromium blog post](https://blog.chromium.org/2021/01/limiting-private-api-availability-in.html) or [this omgUbuntu article](https://www.omgubuntu.co.uk/2021/01/chromium-sync-google-api-removed) for more info.

What fixed the login into Google sites for me was launching Chromium with the Google Chrome API keys from [here](https://gist.github.com/foutrelis/14e339596b89813aa9c37fd1b4e5d9d5) (replace the `...` with the values from that page):
```bash
export GOOGLE_DEFAULT_CLIENT_ID=...
export GOOGLE_DEFAULT_CLIENT_SECRET=...
chromium-browser
```

**Note that this is not officially supported by Google**, and these keys may stop working at any moment.
I think the Raspian package is not built correctly because it does show the possibility to login to sync with Google (suggesting that some Google API keys were entered at built-time), but the sync doesn't actually work.

XECDesign, 2021-12-11:
Yup, I think it's both. The API keys don't work, so when the browser tries to store the session for Google's sites, it quietly fails. Toggling the option I mentioned earlier did fix it for me, so it's at least related.

I have a build I need to test on Monday which has browser sign-in disabled and API keys removed.

XECDesign, 2021-12-15:
This should be resolved now.

slokhorst, 2021-12-15:
Yep, fixed for me too! Thanks!

closed 2021-12-24 by XECDesign

HoldOffHunger, 2022-02-28:
Hi, thanks for detailing this issue!

It's still broken for me when using a brand-new image from the raspberry pi site.  For instance, the first OS here still has this problem: https://www.raspberrypi.com/software/operating-systems/

I think it's the same problem as here, because I solved it the same way: disable Chromium login.  (What is chromium login?  I google it and every page is basically titled, "Can't Login to Chromium.")

Without disabling Chromium login, it's impossible to change google sync settings in Chrome.