RPi-Distro/chromium #22: [Bullseye] Chromium depends on libgtk-3-0

MichaIng, 2021-10-13:
There seems to be a package dependency missing: When trying to start Chromium without `libgtk-3-0` installed, it exits immediately, without any error message in X server logs. Installing this library, resolves it. I also assured that it is not a dependency, as only removing this library (keeping all its dependencies installed) breaks Chromium startup again.

Shouldn't play a role, but it is the same regardless whether fMKS, KMS or legacy graphics API is used.

This is also a dependency of Debian's `chromium` package, but of course that along doesn't mean something as sources are different.

MichaIng, 2021-11-25:
Related: https://forums.raspberrypi.com/viewtopic.php?t=320424

XECDesign, 2021-12-07:
Should be resolved now

MichaIng, 2021-12-07:
Many thanks, yes I can confirm it's solved 👍. Marking this issue as closed.

closed 2021-12-07 by MichaIng