RPi-Distro/chromium #25: [Buster] Chromium fails to start on ARMv6

rxhfcy, 2022-01-13:
This is basically #21, but for Buster. (While it can most definitely be argued that using Chromium on a Raspberry Pi Zero or original RasPi ain't what it used to be, it can still be very usable for showing simple static web pages etc.)

If you already had an existing fully updated old Buster installation, or download 2021-05-07 from the server and update that, Chromium still works fine on a Raspberry Pi Zero (https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-05-28/).

However, if you install the new "Legacy" Buster, it breaks Chromium on ARMv6 ("Illegal instruction") (https://www.raspberrypi.com/software/operating-systems/ currently links to raspios_oldstable_armhf-2021-12-02)

The release notes tell us that "2021-12-03:  * Replaced hardware accelerated Chromium build with Raspbian's stock version"
https://downloads.raspberrypi.org/raspios_oldstable_armhf/release_notes.txt

Here's a list of chromium packages installed on a 2021-05-07 Buster:
```
chromium-browser-l10n/oldstable,now 92.0.4515.98~buster-rpt2 all [installed,automatic]
chromium-browser/oldstable,now 92.0.4515.98~buster-rpt2 armhf [installed]
chromium-codecs-ffmpeg-extra/oldstable,now 92.0.4515.98~buster-rpt2 armhf [installed,automatic]
rpi-chromium-mods/oldstable,now 20210212 armhf [installed]
```
...and here's what's installed on a 2021-12-02 "Legacy" Buster:
```
chromium-common/oldstable,now 90.0.4430.212-1~deb10u1 armhf [installed,automatic]
chromium-sandbox/oldstable,now 90.0.4430.212-1~deb10u1 armhf [installed,automatic]
chromium/oldstable,now 90.0.4430.212-1~deb10u1 armhf [installed]
```
Switching 2021-12-02 back from Raspbian's stock version makes the browser work again.
```
sudo apt remove chromium
sudo apt autoremove
sudo apt install chromium-browser rpi-chromium-mods
```
After that, rather unsurprisingly, the list of 2021-12-02 chromium packages is the same as in 2021-05-07, and Chromium seems to work nicely.
```
chromium-browser-l10n/oldstable,now 92.0.4515.98~buster-rpt2 all [installed,automatic]
chromium-browser/oldstable,now 92.0.4515.98~buster-rpt2 armhf [installed]
chromium-codecs-ffmpeg-extra/oldstable,now 92.0.4515.98~buster-rpt2 armhf [installed,automatic]
rpi-chromium-mods/oldstable,now 20210212 armhf [installed]
```

rxhfcy, 2022-01-13:
For reference, here's what's installed on a fully updated 2021-10-30-bullseye (Chromium currently works!)
```
chromium-browser-l10n/stable,now 95.0.4638.78-rpt5 all [installed,automatic]
chromium-browser/stable,now 95.0.4638.78-rpt5 armhf [installed]
chromium-codecs-ffmpeg-extra/stable,now 95.0.4638.78-rpt5 armhf [installed,automatic]
rpi-chromium-mods/stable,now 20220111 armhf [installed]
```

XECDesign, 2022-01-14:
That's expected. We're not going to maintain Chromium for legacy releases, so instead of pre-installing a version that will accumulate security issues, we switched to the versions provided by Raspbian and Debian. Unfortunately, those versions don't support armv6.

closed 2022-01-14 by XECDesign

bengoerz, 2023-02-28:
> For reference, here's what's installed on a fully updated 2021-10-30-bullseye (Chromium currently works!)
>
> ```
> chromium-browser-l10n/stable,now 95.0.4638.78-rpt5 all [installed,automatic] [...]

Just to clarify, Chromium on current 2023-02-21 Bullseye release does not work on ARMv6 processors (Pi Zero / Zero W / RPi 1).

What's installed on 2023-02-21 Bullseye:
chromium-browser-l10n/stable,now 109.0.5414.112-rpt2 all [installed,automatic]
chromium-browser/stable,now 109.0.5414.112-rpt2 armhf [installed]
chromium-codecs-ffmpeg-extra/stable,now 109.0.5414.112-rpt2 armhf [installed,automatic]
rpi-chromium-mods/stable,now 20220711 armhf [installed]

If you must run Chromium on ARMv6 Pis today, you can install the legacy Buster image and reinstall Chromium (per instructions above to restore to 2021-12-02 version).