RPi-Distro/chromium #26: Right clicking disables compositing hardware acceleration and breaks WebGL

ssesik, 2022-01-16:
Hi
I have found that by right clicking inside chromium window makes chromium disable compositing hardware acceleration and breaks WebGL.
I am using Raspberry Pi 4 and I found that both 32bit and 64bit bullseye rpi os have the same problem.

Here is some screenshot of chrome://gpu
[https://imgur.com/a/BxrBFPb](https://imgur.com/a/BxrBFPb)
[https://imgur.com/a/nC5pv7f](https://imgur.com/a/nC5pv7f)

Log Message of chrome://gpu
`[2151:2151:0116/215031.213655:ERROR:gpu_init.cc(453)] : Passthrough is not supported, GL is egl, ANGLE is
[2151:2151:0116/215122.755496:ERROR:glsurfaceegl.cc(780)] : EGL Driver message (Error) eglMakeCurrent: eglMakeCurrent
[2151:2151:0116/215122.756788:ERROR:skiaoutputsurfaceimplon_gpu.cc(1585)] : Failed to make current.
[2151:2151:0116/215122.758529:ERROR:skiaoutputsurfaceimplon_gpu.cc(1415)] : Failed to make current during initialization.
[2151:2151:0116/215122.773774:ERROR:sharedimagestub.cc(508)] : SharedImageStub: context already lost
[2151:2151:0116/215122.774843:ERROR:raster_decoder.cc(1212)] : RasterDecoderImpl: Context lost during MakeCurrent.
[2151:2151:0116/215122.785022:ERROR:raster_decoder.cc(1212)] : RasterDecoderImpl: Context lost during MakeCurrent.
[2151:2151:0116/215122.785926:ERROR:raster_decoder.cc(1212)] : RasterDecoderImpl: Context lost during MakeCurrent.
[2151:2151:0116/215122.790259:ERROR:sharedimagestub.cc(508)] : SharedImageStub: context already lost
[2151:2151:0116/215122.793783:ERROR:sharedimagestub.cc(508)] : SharedImageStub: context already lost
[2151:2151:0116/215122.797373:ERROR:sharedimagestub.cc(508)] : SharedImageStub: context already lost
[2151:2151:0116/215122.798067:ERROR:sharedimagestub.cc(508)] : SharedImageStub: context already lost
[2151:2151:0116/215122.806899:ERROR:skiaoutputsurfaceimplon_gpu.cc(1585)] : Failed to make current.
[2151:2151:0116/215122.807878:ERROR:skiaoutputsurfaceimplon_gpu.cc(1585)] : Failed to make current.
[2151:2151:0116/215122.809769:ERROR:skiaoutputsurfaceimplon_gpu.cc(1585)] : Failed to make current.`
WebGL before
[image]

WebGL after
[image]

XECDesign, 2022-01-17:
Yup, looks like things are horribly broken.

@jc-kynesim, @naushir, @spl237, I think it crashes and switches to software rendering whenever you open chromium and right click on anything. Bumping up CMA doesn't help. Downgrading to Buster's build of chromium (v92) fixes it.

Is this one for Igalia?

naushir, 2022-01-17:
@itoral and @txenoo, would you be able to have a quick look at this please?

XECDesign, 2022-01-17:
Could be an upstream chromium issue or something I'm doing in my build, so I wouldn't assume it's anything you've done.

txenoo, 2022-01-17:
I can reproduce it. It happens also on shadertoy samples with chromium-browser 95.0.4638.78-rpt5 on bullseye.

It seems that it could be a regression caused by a chromium update, I tested it reinstalling previous chromium and I don't see this behaviour anymore:

 pi@teclas-rpi4-4gb:~$ dpkg -l|grep chromium
ii  chromium-browser                       92.0.4515.98-rpt2                    armhf        Chromium web browser, open-source version of Chrome
ii  chromium-browser-l10n                  92.0.4515.98-rpt2                    all          chromium-browser language packages
ii  chromium-codecs-ffmpeg-extra           92.0.4515.98-rpt2                    armhf        Extra ffmpeg codecs for the Chromium Browser
ii  rpi-chromium-mods                      20210812                             armhf        Raspberry Pi-specific mods to Chromium

I got previous chromium package version from here:

http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-browser_92.0.4515.98-rpt2_armhf.deb
http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-browser-l10n_92.0.4515.98-rpt2_all.deb
http://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/chromium-codecs-ffmpeg-extra_92.0.4515.98-rpt2_armhf.deb

jc-kynesim, 2022-01-17:
I think I know what it is - chrome calculates bits-per-pixel of the underlying renderer in two different places in two slightly different ways in a manner that can give different results.  If it fails to match you get the "Failed to make current" errors I'm seeing. I had a kludge for this in an earlier chrome but I thought the issue had gone away so I think I took it out. I'll double check the status. The particularly wonderful thing about this "feature" was it swapped between KMS/FMKS.  I'm going to assume that if I can only easily (due to sandboxing) get it right for one variation it will be KMS.

jc-kynesim, 2022-01-17:
I have absolutely no idea why a right-click triggers this

jc-kynesim, 2022-01-17:
No... I was right to lose that patch/kludge - if I have it in then we are broken from the start. So more investigation required...

XECDesign, 2022-01-19:
A build with jc-kynesim's fix seems to work and should be in the apt repo now.

ssesik, 2022-01-19:
Yep, it is fixed although there is a warning `WARNING:gl_surface_egl_x11_gles2.cc(116)] : Geometry depth(24) != Config depth(32)` every time I right click. I suppose it is normal.

Thanks

jc-kynesim, 2022-01-19:
I added the warning so it is obvious when the "fix" is doing something - I haven't fixed the underlying bug (and I don't know how to), I've just forced the code to avoid the failing situation.