RPi-Distro/chromium #28: `chromium-browser $website` Always Fails

HoldOffHunger, 2022-02-28:
Hi,

Loading a website with `chromium-browser wikipedia.org`, for instance, will never ever actually load the page, and it will be forever stuck in a network loop.  This is also the case with `chromium-browser --app=https://wikipedia.org` and `chromium-browser --kiosk wikipedia.org`.  Running it within `lxterminal`, similarly, just fails.

On the other hand, `firefox wikipedia.org` loads instantly, fast, and quite conveniently.

XECDesign, 2022-03-09:
I'm unable to reproduce the issue with any of the commands provided.

Is there anything unusual about your setup? Is it an official Raspberry Pi OS image (non-lite)? Any details that would help reproduce the issue would be good.

HoldOffHunger, 2022-03-09:
Hey hey,

I'm using Raspberry Pi OS with desktop here: https://www.raspberrypi.com/software/operating-systems/

Specifically, 2022-01-28-raspios-bullseye-armhf.  I retried it now with Raspiberry Pi 3b+ and a Raspiberry Pi W 0, and both experience the same issue.

I've made a video of the issue: https://www.youtube.com/watch?v=0FpPnX6prZ0

It takes about 45-60 seconds for wikipedia to load.  But then, `chromium-browser wikipedia.org` never loads, ever.  I have 4 minutes of it loading; it kept going for about 20 minutes with the loading icon.  In fact, it's still loading right now, as I'm typing this.  So, I think something's not quite right.

XECDesign, 2022-03-22:
Github is not letting be add a comment to the line: https://github.com/RPi-Distro/pi-gen/pull/602/files#diff-8f8b4cc12fb49987657f96bdd01eefbbaddf18d785059cc0aeb852983ce5e26cR10

Unless I'm missing something, if `proxy_check` fails, this seems to be disabling it regardless of whether `APT_PROXY_FALLBACK` is set.

jordanreedy16, 2022-03-24:
Hey there,

Running into the same issue after installing a new version of Raspbian from scratch. Using firefox for now but I'd like to help get this fixed. 

Hardware: Raspberry Pi 3 Model B Rev 1.2
OS Image: Raspbian GNU/Linux 11 (bullseye)

Below is the stderr log without the --verbose flag.

`libEGL warning: DRI2: failed to authenticate
[1717:1717:0324/123947.652731:ERROR:component_loader.cc(188)] Failed to parse extension manifest.
[1769:1769:0324/123948.162615:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is
[1769:1769:0324/123948.243729:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1717:1826:0324/123949.082228:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1717:1826:0324/123949.084177:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.
GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1717:1826:0324/123949.085547:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
libEGL warning: DRI2: failed to authenticate
[1825:1825:0324/123949.836026:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is
[1825:1825:0324/123949.946281:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1852:1852:0324/123950.029009:ERROR:egl_util.cc(74)] Failed to load GLES library: /usr/lib/chromium-browser/libGLESv2.so: /usr/lib/chromium-browser/libGLESv2.so: cannot open shared object file: No such file or directory
[1852:1852:0324/123950.092159:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1861:1861:0324/123950.142643:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is disabled, ANGLE is
[1717:1751:0324/123953.767139:ERROR:chrome_browser_main_extra_parts_metrics.cc(227)] START: ReportBluetoothAvailability(). If you don't see the END: message, this is crbug.com/1216328.
[1717:1751:0324/123953.767391:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()
[1770:1781:0324/124139.515101:ERROR:ssl_client_socket_impl.cc(995)] handshake failed; returned -1, SSL error code 1, net_error -101
[1770:1781:0324/124139.531698:ERROR:ssl_client_socket_impl.cc(995)] handshake failed; returned -1, SSL error code 1, net_error -101
[1807:4:0324/124216.509957:ERROR:child_thread_impl.cc(345)] Crashing because hung
[0324/124216.530664:ERROR:elf_dynamic_array_reader.h(64)] tag not found
[1770:1781:0324/124216.641807:ERROR:node_controller.cc(585)] Trying to re-add dropped peer 26BD6E47E5FC3D3D.ED40827B2485329
[1772:4:0324/124216.642602:ERROR:node_controller.cc(585)] Trying to re-add dropped peer 26BD6E47E5FC3D3D.ED40827B2485329`

[stderr-nonverbose.log](https://github.com/RPi-Distro/chromium-browser/files/8343756/stderr-nonverbose.log)

My steps to reproduce are 
1. Run `chromium-browser --kiosk --incognito https://www.google.com`
2. Wait until the page says unresponsive, click "wait"
3. Wait until it says unresponsive again, then click "kill"

I have verbose logs I'll attach here too. Same steps as above.
[chromiumbrowser-verbose.log](https://github.com/RPi-Distro/chromium-browser/files/8343741/chromiumbrowser-verbose.log)

Going to keep banging on it in the meantime. Can get more info if need be.

MichaIng, 2022-03-31:
The symptoms look similar to the issue we face: https://github.com/MichaIng/DietPi/issues/5342

Solution is to enable KMS, but on Raspberry Pi OS Bullseye this should be enabled by default. @jordanreedy16 can you verify that you use KMS and not the legacy framebuffer driver, and that as well X11 uses the modesetting driver and not fbdev or fbturbo?

jordanreedy16, 2022-03-31:
Hey @MichaIng , the advice you suggested in the other thread worked like a charm! Specifically,

```
G_CONFIG_INJECT 'dtoverlay=vc4-f?kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"' > /etc/chromium.d/egl
reboot
```

I didn't check to see which drivers were in use prior to the change, but I did do a raspi-config, chose the Full KMS driver and rebooted. That initially did not fix the issue. Further testing indicates that changing the settings in raspi-config makes no difference. It works regardless at this point. Thanks again for the assist!

MichaIng, 2022-03-31:
> I did do a raspi-config, chose the Full KMS driver and rebooted

`raspi-config` adds the same line to `config.txt`, but `--use-gl=egl` is not added to Chromium config of course, so probably this made the difference. It makes sense anyway to use EGL in this combination.

AFAIK there is not really much downside on any RPi model when using KMS, it is the default on RPi OS anyway, but I still wonder whether it is intended or expected that Chromium does not work with framebuffer driver anymore, or whether framebuffer driver support can be re-enabled somehow.

XECDesign, 2022-04-04:
Maybe I've missed some detail, but I'm still having trouble reproducing this on any official image with kms disabled.

taoteh1221, 2022-04-05:
> Maybe I've missed some detail, but I'm still having trouble reproducing this on any official image with kms disabled.

If you have a raspi v3B+, flash it's microsd with either 32bit or 64bit 2022-1-28 Desktop Raspi Official OS, simply run apt update;apt upgrade (don't do anything else), then try a REMOTE SSH login and type these commands:

```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Apr  5 18:03:19 2022
pi@raspberrypi:~ $ FIND_DISPLAY=$(w -h $USER | awk '$3 ~ /:[0-9.]*/{print $3}')
pi@raspberrypi:~ $ DISPLAY=$FIND_DISPLAY
pi@raspberrypi:~ $ export DISPLAY=$FIND_DISPLAY
pi@raspberrypi:~ $ chromium-browser https://google.com
```

You should get output in the terminal similar to this:

```
libEGL warning: DRI2: failed to authenticate
[1600:1600:0405/180536.246197:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is 
[1600:1600:0405/180536.403281:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1549:1652:0405/180537.419033:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1549:1652:0405/180537.421956:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1549:1652:0405/180537.424881:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
libEGL warning: DRI2: failed to authenticate
[1651:1651:0405/180538.883872:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is 
[1651:1651:0405/180538.960905:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1695:1695:0405/180539.020919:ERROR:egl_util.cc(74)] Failed to load GLES library: /usr/lib/chromium-browser/libGLESv2.so: /usr/lib/chromium-browser/libGLESv2.so: cannot open shared object file: No such file or directory
[1695:1695:0405/180539.034505:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1705:1705:0405/180539.077319:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is disabled, ANGLE is 
[1705:1705:0405/180539.090552:ERROR:sandbox_linux.cc(377)] InitializeSandbox() called with multiple threads in process gpu-process.
[1549:1598:0405/180541.277733:ERROR:chrome_browser_main_extra_parts_metrics.cc(227)] START: ReportBluetoothAvailability(). If you don't see the END: message, this is crbug.com/1216328.
[1549:1598:0405/180541.277954:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()
```

On the attached display, output should be similar to this (just hangs, NEVER loads google.com):

[image]

taoteh1221, 2022-04-05:
> The symptoms look similar to the issue we face: [MichaIng/DietPi#5342](https://github.com/MichaIng/DietPi/issues/5342)
> 
> Solution is to enable KMS, but on Raspberry Pi OS Bullseye this should be enabled by default. @jordanreedy16 can you verify that you use KMS and not the legacy framebuffer driver, and that as well X11 uses the modesetting driver and not fbdev or fbturbo?

@MichaIng that fix works great, thank you! :) For anyone else reading this thread, as a regular user in the terminal (or bash script), just add sudo after the pipe:

```
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"' | sudo tee /etc/chromium.d/egl
```

MichaIng, 2022-04-06:
@taoteh1221 
```
[1651:1651:0405/180538.883872:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is 
[1651:1651:0405/180538.960905:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1695:1695:0405/180539.020919:ERROR:egl_util.cc(74)] Failed to load GLES library: /usr/lib/chromium-browser/libGLESv2.so: /usr/lib/chromium-browser/libGLESv2.so: cannot open shared object file: No such file or directory
[1695:1695:0405/180539.034505:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[1705:1705:0405/180539.077319:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is disabled, ANGLE is 
```
Actually it looks like EGL is tried to be used already at first, but failing due to missing `libgles2` (`apt install libgles2`), so it falls back to no GL. The more strange it is that passing the flag solves it. Probably it overrides the otherwise automatic fallback.

On my latest test on RPi 2, even without explicitly defining `--use-gl=egl`, EGL was used automatically, if KMS is enabled, `libegl1` and `libgles2` installed. I was already [thinking](https://forums.raspberrypi.com/viewtopic.php?p=1984070#p1984070) whether it makes sense to add `libgles2` (`libegl1` is pulled in already) as package dependency, but couldn't find the right way to do this upstream, i.e. where hardware/build flag specific package dependencies are handled.

taoteh1221, 2022-04-06:
That said, MAYBE `apt install libgles2` modifies `/etc/chromium.d/egl` when installed? I don't know, busy patching up my related software for users having issues because of this RN.

MichaIng, 2022-04-06:
> That said, MAYBE apt install `libgles2` modifies `/etc/chromium.d/egl` when installed?

Definitely not. This is just a random filename (chosen to match the purpose), you can choose any other or add the flag to any other file within this directory. And no library is manipulating any other software's configuration files.

But still confusing. On DietPi we add the following flags:
```sh
--no-sandbox --test-type --disable-smooth-scrolling --disable-low-res-tiling --enable-low-end-device-mode --num-raster-threads=$(nproc) --disable-composited-antialiasing
```
No idea whether these somehow trigger EGL.

However, while we do not fully understand these little differences, it doesn't matter much. With KMS enabled, using EGL with GLES makes sense, respectively is required, so good to add the flag explicitly.

The more important question is whether it is intended and unavoidable to use KMS from Chromium now. Generally it makes sense and should provide better performance, but there are still some applications relying on features available via legacy framebuffer driver only, or other reasons why user might still want to use it, while keep using Chromium.

taoteh1221, 2022-04-06:
> > That said, MAYBE apt install `libgles2` modifies `/etc/chromium.d/egl` when installed?
> 
> Definitely not. This is just a random filename (chosen to match the purpose), you can choose any other or add the flag to any other file within this directory. And no library is manipulating any other software's configuration files.
> ...
>The more important question is whether it is intended and unavoidable to use KMS from Chromium now. Generally it makes sense and should provide better performance, but there are still some applications relying on features available via legacy framebuffer driver only, or other reasons why user might still want to use it, while keep using Chromium.

Good to know. I'm using jquery transitions / animations, which seem to work MUCH better with hardware accelerated browsers. `firefox-esr` doesn't provide that, and it shows (via stuttering etc). `chromium-browser` seems to be the only way to go on a pi device, although `epiphany-browser` does OK.

XECDesign, 2024-09-18:
Is this still an issue?

jswolf19, 2024-09-18:
@XECDesign We're replacing a pi that uses this fix with one using Bookworm in the near future. I'll let you know if the fix is still necessary or not for our use case when we do.

jswolf19, 2024-09-23:
@XECDesign The problem still occurs on a Pi 3B using RaspberryPi OS Bookworm and 
chromium/stable,now 1:128.0.6613.137-1~deb12u1+rpt3 arm64

XECDesign, 2024-09-23:
Ah, that's a shame. Thanks for checking. I have just never seen the issue myself and have no idea how to reproduce it.

You're just taking the latest 64-bit desktop Raspberry Pi OS image, running chromium through the cli and it's not working?

jswolf19, 2024-09-24:
I'm using the Raspberry Pi 64-bit OS image that Imager provides, did an `apt full-upgrade` followed by an `apt autoremove`.
Then I have the following command in ~/.config/autostart.

```bash
/usr/bin/chromium-browser --app=<private url> --start-fullscreen --force-device-scale-factor=1.00
```

The page loads, then fetches data, which refreshes the view. The initial load seems to work, but when the data gets fetched instead of the view updating, it goes white.

I'll see about getting a repro with static files.

jswolf19, 2024-09-24:
@XECDesign https://towa-japan.github.io/RPi-Chromium-28/ exhibits the issue I'm having (on a raspberry Pi 3B).

Using
```bash
/usr/bin/chromium-browser
```
I get a blank page, and the following error output:
```
[4756:4813:0924/142934.643964:ERROR:v4l2_utils.cc(572)] Could not open /dev/video10: Permission denied (13)
[4756:4813:0924/142934.648569:ERROR:v4l2_utils.cc(572)] Could not open /dev/video10: Permission denied (13)
[4756:4813:0924/142938.718557:ERROR:v4l2_utils.cc(572)] Could not open /dev/video10: Permission denied (13)
```

Using
```bash
/usr/bin/chromium-browser --use-gl=egl
```
I get the expected page, and the following error output:
```
[4552:4552:0924/142801.710514:ERROR:gl_factory.cc(98)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[4552:4552:0924/142801.722957:ERROR:viz_main_impl.cc(179)] Exiting GPU process due to errors during initialization
[4591:4591:0924/142802.792791:ERROR:gl_factory.cc(98)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[4591:4591:0924/142802.812497:ERROR:viz_main_impl.cc(179)] Exiting GPU process due to errors during initialization
[4610:4610:0924/142803.024508:ERROR:gl_factory.cc(98)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[4610:4610:0924/142803.046041:ERROR:viz_main_impl.cc(179)] Exiting GPU process due to errors during initialization
[4617:4617:0924/142803.165824:ERROR:gl_factory.cc(98)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[4617:4617:0924/142803.180070:ERROR:viz_main_impl.cc(179)] Exiting GPU process due to errors during initialization
```

XECDesign, 2024-09-25:
I'm seeing the reported behaviour under X11 and Wayfire, but it seems to work as expected under labwc. We'll be switching to labwc by default soon, so it may be worth checking if it solves your problem too. In raspi-config, you can go to advanced -> wayland and switch to labwc.

jswolf19, 2024-09-25:
Is that true for a 3B? I know the plan is to support older hardware on wayland, but last I checked, it'll only run on a pi4 or newer.

XECDesign, 2024-09-26:
It is definitely true for the 3B and I believe across the board for all devices.

jswolf19, 2024-09-26:
I'll try it out and get back to you, then.

jswolf19, 2024-09-27:
I was able to display the private url without adding any extra flags to the chromium-browser commandline. I do, however, run into the [issue](https://github.com/any1/wayvnc/issues/318) of a grayscreen when using vnc, so I'll wait a bit before making this change in production use.

XECDesign, 2024-10-11:
Closing since I believe this and the vnc issues are both resolved.

closed 2024-10-11 by XECDesign

jswolf19, 2024-10-11:
Sorry, yes the vnc issue is resolved as discussed at the linked [issue](https://github.com/any1/wayvnc/issues/318)