RPi-Distro/chromium #30: Sometimes uBlock Origin doesn't work the first time after clicking "Restore Session" after an unexpected shutdown

HoldOffHunger, 2022-04-05:
Hi, all,

MACHINE: Raspiberry Pi 3b+

PROBLEM: Sometimes uBlock Origin doesn't work first time after clicking "restore session."

_Sometimes_ uBlock origin is installed (and I haven't changed any of the configs with the plugins), and have done no modifications to the system when using a brand new image.  If I'm watching Youtube, and power is unexpectedly cut from the Pi, when I reboot and re-open Chromium, I get the dialogue, "Chrome shutdown unexpectedly.  Restore?"

If (very quickly) I click "YES", the last youtube page will reload, but it'll begin by playing an ad!  It clearly feels like it happens when the browser is just starting up.  After the first initial ad, no further ads will play, it's just the first page right after you click "Restore."

REPLICATION:

I would say this happens about 20% of the time, which makes this very frustrating.

I have no idea what system logs could at all be useful in this situation.  And I admit, it's very weird behavior to try and reproduce.

XECDesign, 2022-04-07:
I don't think that's a problem specific to our chromium build. AIUI, that's just chromium loading a page before the extension is loaded.

HoldOffHunger, 2022-04-07:
That seems reasonable.  I guess we can close this.  I was only reporting it here because I'm regularly noticing it on my pi but have never once seen this happen on a PC with Chrome.

It also looks like they don't allow submitting issues to their project.  Excellent.

closed 2022-04-07 by HoldOffHunger