RPi-Distro/chromium #33: Chromium 92 issue

ChrisNic, 2022-05-24:
i have a pi3b running buster using Chromium 92. it recognizes only my usb webcam not the picam. if i open caprine it sees bot camera's. usb webcam has bad quality but built in mic. i can only video chat in messenger but would like to use other applications ie. telegram, meet, etc.
[image]
[image]

XECDesign, 2022-05-25:
Duplicate of #23?

XECDesign, 2022-05-25:
Also, we don't support v92. We currently ship v101. If you're on buster, you're encouraged to switch to Debian's `chromium` package (rather than `chromium-browser`).