RPi-Distro/chromium #34: SVG Flickering

KLIIIMENT, 2022-06-24:
Hello everyone,

I observe the strange behavior by the showing of the svg plot. The svg-plot was created with Plotly.
After zooming in I can observe flickering. 
I checked svg and it stays constant without any changes.
In the invisible(overlayed through layer with plot) background I have 2 HTML fields which updates every second. 
If I disable the updating of the 2 fields then I don't see the flickering. Instead of sometimes I can observe  the corrupt svg plot.

- Chromium started in kiosk mode with X Server

P.S.: I checked the issue with other browsers and it doesn't appear
In attached archive you can find the index.html and plotlytest.js files, so you can easy reproduce the bug.

**CHROMIUM VERSION:**
Chromium 101.0.4951.57 Built on Raspbian , running on Raspbian 11

**HARDWARE**:
Raspberry Pi 3 Model B Plus Rev 1.3

**OS:**
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[ChromiumBug.zip](https://github.com/RPi-Distro/chromium-browser/files/8974007/ChromiumBug.zip)

https://user-images.githubusercontent.com/108117025/175490190-c307adb6-9317-4a88-8fa7-1f95c03c9bff.mp4

XECDesign, 2024-09-18:
Is this still an issue in bookworm?

KLIM-T, 2024-09-19:
Hi @XECDesign, I did not encounter this issue in bookworm.

closed 2024-09-19 by KLIM-T