RPi-Distro/chromium #35: Hardware accelerated Chromium on CM3+ does not render correctly

JinShil, 2022-07-25:
For context, please see https://forums.raspberrypi.com/viewtopic.php?t=337558

Pi Hardware: CM3+
OS: Latest 32-bit Raspberry Pi OS Bullseye downloaded today
Browser version: Chromium 101.0.4951.57 Built on Raspbian , running on Raspbian 11

View the graphics demo at https://pixijs.io/examples/#/demos-basic/blendmodes.js on any desktop PC as a reference for what it should look like.

On a CM3+, using `dtoverlay=vc4-kms-v3d`, and with Glamor enabled, the demo looks like this:

[image]

The speed at which the sprites move indicates the hardware acceleration pipeline is being utilized, but the graphics are just not rendered correctly.

Disabling Glamor, and adding command line arguments `--use-gl=angle --use-angle=gl-egl --use-cmd-decoder=passthrough` seems to work around the issue.  The workaround does not work if Glamor is enabled, which is quite curious.

The demo functions correctly on the CM4.

XECDesign, 2022-07-26:
@naushir

jasuarez, 2022-08-29:
This is [fixed in Mesa upstream](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/18122).

XECDesign, 2022-08-30:
@spl237 Can this be backported to our version?

JinShil, 2023-10-27:
This is no longer an issue with the latest OS releases.

closed 2023-10-27 by JinShil