RPi-Distro/chromium #36: Electron hardware acceleration

rhysmorgan134, 2022-12-05:
Hi Apologies if this is completely the wrong place to ask! Over in the electron repo there is an issue where gpu acceleration with in electron does not work. Someone managed it by rebuilding with some of your patches. Electron have stated that if the flags are up streamed to chromium then electron will pull them in automatically.

Is this something that’s feasible or a complete no go?

Link to electron issue

https://github.com/electron/electron/issues/34825

XECDesign, 2022-12-12:
I'm not aware of any plans to upstream this. I don't think there's much hope of anyone from google being interested in reviewing the patch.

theofficialgman, 2022-12-17:
I am aware of many many projects interested in getting electron and vanilla chromium working on the pi systems.

Upstreaming the basic patches so that electron and vanilla chromium can be run at all should be top priority. What myself and other users are seeing now is MESA-LOADER errors (when using fkms or kms) and then no output.
```
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open kms_swrast: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: Permission denied (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: Permission denied (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load swrast driver
```

unic8s, 2023-04-30:
Try this with your electron app
`"strace -f node_modules/.bin/eletron release/app`

Rerferring to that post https://forums.raspberrypi.com/viewtopic.php?t=338778#p2029078 there are more strange underlying 
 things going on.

Many projects rely on the platform and electron software.

closed 2023-05-02 by XECDesign

szgerg, 2023-06-03:
Hi all,
Why is this issue closed? I have exactly the same issue.
What is the solution?
Thanks