RPi-Distro/chromium #38: Freezing the machine under Rapsbian (64-bit version)

atimmernans, 2023-08-08:
I can reproduce it at will but when browsing articles and the conversions at the end of it from the warzone, like this one:
https://www.thedrive.com/the-war-zone/french-scalp-eg-cruise-missiles-officially-in-use-in-ukraine

freeze the machine (a rapsberry pi-4 with 4 GB or RAM and an SSD drive).

When expending, scrolling there is often high CPU/disk activity due to high number of linked images/ videos/ thumbnailed links but normally it dies down as soon as they have finished loading. However once or twice a week, the activity doesn't stop,  the machine stops responding to mouse clicks or to the keyboard and after some minutes, even the mouse stops moving.

XECDesign, 2023-08-08:
Having trouble reproducing the issue here.  The page loaded with no issues. I've opened around 15 tabs with random links from that site, scrolled around while switching tabs and while it does pin all core at 100%, the system seems stable.

andrum993, 2023-08-15:
Hi folks. Not sure if this bug or the Linux kernel bug is the correct one to add this to, but just to add some more information, there is a form thread I have spotted that provides another test case, and a bit more info from another person who has experienced this crash. The forum thread is at https://forums.raspberrypi.com/viewtopic.php?t=354939.

popcornmix, 2023-08-15:
As in the [linked issue](https://github.com/raspberrypi/linux/issues/5574) can you see if adding `gpu_mem=256` avoids the issue?
(this isn't a fix - it is just to confirm the issue is related to multiple hw video decoders exhausting gpu memory).

atimmernans, 2023-08-16:
I will try with 256MB of GPU but note that in the meantime I have turned hardware acceleration off and so far have not encountered the issue.

sudo-splinter-cell, 2023-12-14:
Just throwing this out there. If you are on a Pi and having system freezes while browsing the web on Chromium or Firefox, the issue is usually your ram being filled up. Unfortunately, those web browsers are really very bad at managing their ram usage and they just throw in anything into the ram and not care about it. Modern systems as we know have huge amounts of ram which is ok but for us with 4 or 8 gigs of ram, the answer is "increasing the swap space".

[https://pimylifeup.com/raspberry-pi-swap-file/](url)

So follow along the instructions there and you will be fine. I suggest you to have at least "half" the amount of your ram as swap space. Also important is that the drive in which you have your swap file should be a relatively fast one, like an SSD or a fast usb stick. SD card as swap space will hamper the performance quite a lot.

XECDesign, 2024-09-11:
Closing as stale. The chromium package has changed a lot since this was opened and there haven't been any updates on the issue.

closed 2024-09-11 by XECDesign