RPi-Distro/chromium #40: Mouse cursor size resets to "small" while cursor is over Chromium-browser window (on wayland)

sudo-splinter-cell, 2023-12-08:
Chromium-browser does not obey global settings when it comes to mouse cursor size and mouse cursor always defaults to "small" when it is over Chromium window. This happens only on wayland. In modern times with large and high-dpi screens, a small mouse cursor is almost impossible to see.

The debian Chromium package does not have this issue.

Mouse cursor size does not change in any other app window that i opened during my experience of multiple weeks. Every other app actually obeys the global setting in that regard. So it is only the Chromium-browser.

sudo-splinter-cell, 2023-12-20:
https://github.com/RPi-Distro/chromium-browser/assets/79761284/787baafe-3ec5-496b-bad5-4814d4d3eb32

sudo-splinter-cell, 2024-01-30:
This is still an issue.

XECDesign, 2024-10-16:
If I run under wayfire and use the appearance settings app and change the cursor to large, it's working as expected. I assume that means the issue has been resolved by the switch to Debian's packaging or an update to Raspberry Pi OS.

Happy to re-open if somebody is still seeing this and can provide steps to reproduce the issue.

closed 2024-10-16 by XECDesign