RPi-Distro/chromium #45: Bug: Google account requires sign-in everytime I open browser

hubby4, 2024-09-11:
Since the Raspberry Pi OS (64-bit) package manager updated to this new version (128.0.6613.119)  of Chromium, I can no longer stay signed in to my google account.

What I tried:

1) I first tried clearing all cache, cookies, and sign-in data and restart.
2) When that didn't work, I deleted my profile.
3) When that didn't work, I used sudo apt purge chromium, deleted the ~/.config/chromium folder, and reinstalled chromium.

None of these worked. Sign-in works fine with Firefox.

XECDesign, 2024-09-12:
@cillian64 @jc-kynesim

cillian64, 2024-09-12:
Thanks for the bug report.  As a workaround, disabling "Settings" -> "sync and google services" -> "allow chromium sign-in" makes the problem go away for me.  We'll try to get this fixed in a patch release soon.

commit 0ef3128c (2024-09-12): Use Ubuntu's browser sign-in disable-by-default patch instead of Debian's

Fixes #45