RPi-Distro/chromium #46: very first command ./debian/rules get-orig-source doesn't work

vasylenkoserhii, 2024-09-12:
When one tried to follow README.md for building, after cloning this repository:
`git clone --branch pios/bookworm https://github.com/RPi-Distro/chromium.git
cd chromium`
execution of 
`./debian/rules get-orig-source`
gives error 
`wget -nv --show-progress -c https://gsdview.appspot.com/chromium-browser-official/chromium-1:128.0.6613.119.tar.xz -O ../chromium-1:128.0.6613.119.tar.xz
http://commondatastorage.googleapis.com/chromium-browser-official/chromium-1:128.0.6613.119.tar.xz:
2024-09-12 02:43:19 ERROR 404: Not Found.
make: *** [debian/rules:247: get-orig-source] Error 8
`
because `version` read from `debian/changelog` `1:128.0.6613.119` contains extra `1:` at the start.

XECDesign, 2024-09-12:
I am currently taking the orig tarball from debian's repo instead. I don't know whether `get-orig-source` does anything to ensure that the resulting tarball of a given version always ends up with the same checksum.

But yes, it should be fixed to strip out the epoch from the version string.

XECDesign, 2024-09-12:
Hopefully that fixes it (worked for me). And yes, the tarball it generated was identical to debian's so I am a bit more comfortable using get-orig-source now.

commit b40c6802 (2025-07-31): d/r: Use DEB_VERSION_UPSTREAM from pkg-info.mk to determine upstream version

Fixes #46
commit 611745eb (2025-08-26): d/r: Use DEB_VERSION_UPSTREAM from pkg-info.mk to determine upstream version

Fixes #46
commit 44ba94e8 (2025-09-01): d/r: Use DEB_VERSION_UPSTREAM from pkg-info.mk to determine upstream version

Fixes #46