RPi-Distro/chromium #47: HW acceleration features are gone

sudo-splinter-cell, 2024-09-12:
After the latest updates which included the name change of chromium-browser (i think it will be called chromium from now on), i have lost hw acceleration, completely.

[image]

This causes a huge performance slump and causes 3d web pages like Google Earth/Maps not being rendered properly.

I have reverted to Chromium version 125 and now I have hw acceleration.

[image]

This is a huge issue and i hope somebody looks into it.

XECDesign, 2024-09-12:
Are you running under X or wayland?

sudo-splinter-cell, 2024-09-12:
X11

XECDesign, 2024-09-12:
Last I heard that was a wontfix. Switching to labwc should fix it.

sudo-splinter-cell, 2024-09-12:
Why was it broken in the first place? Why do we need to switch to wayland, wayland brings a less responsive desktop experience with it. I hate wayland. Is this a Chromium decision or is this a RPI decision to remove hw acceleration from it under x11? I don't understand this. What was the necessity for doing this?

sudo-splinter-cell, 2024-09-13:
Yeah thanks for the help. I am aware of most of what you say here. I had the compositor disabled a long time ago and i actually "allow" screen tearing (at the cost of vsync), because i got used to my desktop being highly responsive, since windows 98 times, especially the mouse cursor. That mouse cursor responsiveness in wayland is probably just acceptable for most people, who are accustomed to touch screens or who just use high performance machines as their desktop which can provide high frame rates even under load. RPI 5 is not that kind of a machine. So "following the linux trend" in some cases maybe not the wise choice. I have tried to get used to it a couple times and every time i feel like i am "dragging" the actual mouse cursor along the screen with my mind's eye, it is a frustrating experience. Trying to micro-manage mouse movements just to get the cursor to where it should be. And as you stated, because the frame rate is fixed, sometimes mouse cursor becomes "less visible" while moving, especially under gpu load and then i am putting more effort to follow where it is and where it will be

These things might sound like childish complaints but if you use RPI-5 as a desktop, your experience will be the same. Some people might tolerate it, some might not. We are all different.

theofficialgman, 2024-09-13:
> PS: I have already filed bug reports in regard to that "mouse cursor responsiveness" issue to KDE, RPI, Wayland, anywhere that is relevant, a long time ago. I feel like it is not being taken very seriously, as because it affects only us Pi users.

I see no such issue reported regarding "mouse cursor responsiveness" on  https://bugs.kde.org/

> And this is why i think it is not a great idea to adopt wayland in slow hardware like ours. At least until some things are taken care of.

Your complaints to RPI LTD are unfounded about wayland. As I said, each display server implementation is different. Complaining about your KDE Plasma wayland experience to RPI LTD will go nowhere because its not their software and they have no control over it. RPI LTD builds and patches wayfire and labwc (now the preferred compositor) so if you have issues there then you can complain. Otherwise it is pointless.

> There was some rumor about making the mouse cursor have a different layer of render to make it highly responsive even when other parts of the system are not, but i am yet to see it.

Yes this is resolved with https://invent.kde.org/plasma/kwin/-/commit/4c397a95263d9c8e73d536ab1c2ac2caaeec60e2 (the commit linked resolves https://bugs.kde.org/show_bug.cgi?id=472663) (all Plasma 6 releases). On Kwin the cursor was always rendered on a separate plane however they were always synchronized to each other.  So if the desktop rendering was not hitting the frame target, the frame would be delayed, and the cursor would thus feel sluggish. Now (Plasma 6) they are independent and even if the primary plane is not done by the frame target (and thus needs to be delayed by a frame) the mouse position will still be updated.

If you have trouble still after than (ie: such as on a distro that ships Plasma 6) this is again an issue you should file on  https://bugs.kde.org/ . You will need to provide evidence though (ie: high framerate capture of your physical mouse motion with your monitor in the background under X11 and Wayland).

sudo-splinter-cell, 2024-09-13:
Yeah i am currently downloading Fedora Rawhide which claims to have the latest (beta) version of KDE Plasma. I will also test the latest RPIOS desktop. I had the same issues with RPIOS default desktop before when they first switched to wayland but let's see if anything has changed then.

theofficialgman, 2024-09-13:
> If you have trouble still after than (ie: such as on a distro that ships Plasma 6) this is again an issue you should file on https://bugs.kde.org/ . You will need to provide evidence though (ie: [...]

I've done this (on my Pi5 on labwc and PC on Wayland Plasma 6) and measured no difference compared to the baseline for my connected monitor (2-3 frames). The monitors (as tested by a 3rd party) input lag should be 1 frame. Now if the GPU load on the Pi5 is maxed out (due to some other process) then the entire labwc desktop drops frames (including the mouse) but that is likely because on labwc the mouse and primary plane rendering are tied together. Maybe thats not the case and its actually an issue with the hardware cursors implementation but I would expect that to affect both wayland and x11 (since both hardware cursor implementations rely on the same underlying hardware for functionality).

sudo-splinter-cell, 2024-09-13:
You can imagine even the lightest webpages with a little bit of graphics will push the Pi to its limits, i mean even Spotify's main page for example, it is not an easy page to render for the Pi. So to get to that point that you mention is not really very hard for the Pi.

XECDesign, 2024-09-16:
Closing since OP deleted their account and the discussion got a bit derailed anyway.

No, we didn't intentionally break HW 3D acceleration on X11, but I don't know much much than that.

closed 2024-09-16 by XECDesign

XECDesign, 2024-09-16:
Just found that launching with `--use-angle=gles` makes WebGL work quite well.

And judging by `debian/patches/rpi/0012-Init-Prefer-GLES-to-everything-else-on-init.patch`, that should be the default behaviour, maybe I could take a look at what's happening there later.

XECDesign, 2024-09-18:
Should be fixed in today's release.