RPi-Distro/chromium #48: The browser crashes whenever selecting(highlighting) and then right-clicking some content

edgerunning, 2024-10-09:
It used to work fine until an update about a month ago. Now, whenever I highlight some content and then right-click, the browser crashes. I don't know what caused this, but I found out that if I disable graphic hardware acceleration in the browser settings, it works fine and never crashes. It used to work no matter whether hardware acceleration was enabled or not.
I am using the browser on Wayland.

This is the error that occurs when it crashes.
[2338756:2338756:1010/033630.844794:ERROR:wayland_event_watcher.cc(47)] libwayland: xdg_wm_base#15: error 4: wrong configure serial: 180291

[1010/033630.856032:ERROR:elf_dynamic_array_reader.h(64)] tag not found

https://github.com/user-attachments/assets/f0b015de-9f5b-442c-ad39-d2bcbe75e8a6

XECDesign, 2024-10-16:
Are you still seeing this with .100? Which pi are you using and is there a reliable way to reproduce the issue?

On a pi5 running the latest version under wayfire, if I do the same thing as in your video, it works as expected.

edgerunning, 2024-10-16:
> Are you still seeing this with .100?

Yes, I updated to the latest `1:129.0.6668.100-1~deb12u1+rpt1`, but the problem still persists. I also tried labwc, and the issue exists there too, not just with Wayfire.

> Which pi are you using and is there a reliable way to reproduce the issue?

I am using pi 5.
If Under labwc, I run the command `wlr-randr --output HDMI-A-1 --scale 2` to scale it to 2x; you may need to specify a different output. If under Wayfire, I set the scale factor in ~/.config/wayfire.ini. like 
```
[output:HDMI-A-1]
mode = 3840x2160@60000
position = 0,0
transform = normal
scale = 2.0
```
In Chromium, if you open a website, select a clickable hyperlink, and then right click, it crashes.
The problem doesn't occur if you either turn off the '`Use graphics acceleration when available`' option or don’t scale to 2x.

XECDesign, 2024-10-16:
Tried under labwc with `wlr-randr --output HDMI-A-1 --scale 2`, still not seeing the problem. We've still got some difference somewhere.

closed 2024-11-04 by XECDesign