RPi-Distro/chromium #50: Draw call returned Invalid argument

2800GT, 2024-10-14:
I'm running a Moodeaudio build on Raspberry Pi-3B 1.2 1GB with attached DSI LCD display. 
Some elements of the user interface are not being displayed on the screen.
Checking the logs shows the error message "moode2 xinit[11847]: Draw call returned Invalid argument.  Expect corruption."

RaspiOS:       12.7 Bookworm 64-bit
Linux Kernel:  6.6.47 64-bit

Chrome package
```
hi  chromium                             1:129.0.6668.100-1~deb12u1+rpt1        arm64        web browser
hi  chromium-browser                     1:129.0.6668.100-1~deb12u1+rpt1        all          transitional dummy package
hi  chromium-common                      1:129.0.6668.100-1~deb12u1+rpt1        arm64        web browser - common resources used by the chromium packages
hi  chromium-sandbox                     1:129.0.6668.100-1~deb12u1+rpt1        arm64        web browser - setuid security sandbox for chromium
hi  rpi-chromium-mods                    20240904                               arm64        Raspberry Pi-specific mods to Chromium
```

xserver version
```
ii  x11-xserver-utils                    7.7+9+b1                               arm64        X server utilities
ii  xserver-common                       2:21.1.7-3+rpt3+deb12u7                all          common files used by various X servers
ii  xserver-xorg                         1:7.7+23                               arm64        X.Org X server
ii  xserver-xorg-core                    2:21.1.7-3+rpt3+deb12u7                arm64        Xorg X server - core server
ii  xserver-xorg-input-all               1:7.7+23                               arm64        X.Org X server -- input driver metapackage
ii  xserver-xorg-input-libinput          1.2.1-1+b1                             arm64        X.Org X server -- libinput input driver
ii  xserver-xorg-input-wacom             1.1.0-1                                arm64        X.Org X server -- Wacom input driver
ii  xserver-xorg-legacy                  2:21.1.7-3+rpt3+deb12u7                arm64        setuid root Xorg server wrapper
ii  xserver-xorg-video-all               1:7.7+23                               arm64        X.Org X server -- output driver metapackage
ii  xserver-xorg-video-amdgpu            23.0.0-1                               arm64        X.Org X server -- AMDGPU display driver
ii  xserver-xorg-video-ati               1:19.1.0-3                             arm64        X.Org X server -- AMD/ATI display driver wrapper
ii  xserver-xorg-video-fbdev             1:0.5.0-2                              arm64        X.Org X server -- fbdev display driver
ii  xserver-xorg-video-nouveau           1:1.0.17-2                             arm64        X.Org X server -- Nouveau display driver
ii  xserver-xorg-video-radeon            1:19.1.0-3                             arm64        X.Org X server -- AMD/ATI Radeon display driver
ii  xserver-xorg-video-vesa              1:2.5.0-1+b1                           arm64        X.Org X server -- VESA display driver
```
Problem was not present on previous Moode version running chromium browser 126.
Downgrading/reverting back to chromium browser 126 fixes the problem
Adding the switch --disable-gpu to the chromium-browser invocation in /home/user/.xinitrc also appears to remove the problem

moodeaudio, 2024-10-14:
Ok, thanks.

Are there any performance issues when just using --disable-gpu flag?

XECDesign, 2024-10-18:
We'd need steps to reproduce the issue on a standard Raspberry Pi OS image.

moodeaudio, 2024-10-18:
Our OS image is made with pi-gen but the moode-player deb package can be installed directly on fresh Bookworm Lite.

By default we use the latest versions of packages in our build for example chromium and xorg but certain packages that we maintain in our Cloudsmith repo are a specific version. The link below contains the list.
https://github.com/moode-player/imgbuild/blob/main/moode-cfg/stage3_01-moode-install_01-packages

Steps to install moode-player on Lite
```
1. Make a fresh Bookworm Lite 64 bit image with Pi Imager
- Specify hostname "moode" for convenience
- Setup a userid/password
- Enable SSH

3. Boot up on Pi4/5 to take advantage of faster processing
sudo apt update && apt list --upgradable
sudo apt -y upgrade
sudo reboot

4. Install links to moode Cloudsmith repo
curl -1sLf \
'https://dl.cloudsmith.io/public/moodeaudio/m8y/setup.deb.sh' \
| sudo -E distro=raspbian codename=bookworm bash

5. Install moode-player package
sudo apt -y install moode-player=9.1.3-1moode1
# Coffee break...
sudo reboot

6. Moode startup sequence will auto-reboot once more to finalize the installation.

7. ssh userid@moode.local

8. Open the WebUI
http://moode.local
```

Test
```
1. Put uSD Card in Pi3B/3B+
2. Connect HDMI display
3. Boot up
4. Open WebUI (http://moode.local)
5. Menu (m) > Configure > Peripherals
6. Turn on Local Display (the round knob sliders in the middle will not be visible)
7. systemctl status localui
Oct 18 17:23:49 moode9-lite xinit[1980]: Draw call returned Invalid argument.  Expect corruption.
8. Menu (m) > Configure > Peripherals
9. Turn on "Disable GPU" (this adds the flag --disable-gpu to ~/.xinitrc)
10. Knob sliders will appear, there should be no draw call error in systemctl status localui
```

XECDesign, 2024-10-23:
Ha, that's not quite what I meant.  If it involves running a 500 line script and installing packages from third party repos, I don't think it can be called a standard Raspberry Pi OS image anymore.

I tried to do something equivalent:

1) Flashed the lite arm64 image and booted on a 3B with a DSI display
2) Installed xinit, chromium, openbox and xterm.
3) Added a `.xinitrc` file to run `xterm & openbox`
4) Ran startx and launched chromium from the terminal window
5) Tried a few sites and everything rendered as expected.

moodeaudio, 2024-10-23:
Ok yes a generic repro. I'll give your steps a try and see if there are any moode rendering issues.

XECDesign, 2024-10-31:
Any luck?

moodeaudio, 2024-10-31:
Not exactly like your test config but one of our devs created a bare bones system consisting of Pi3 + 64b Lite + chromium + x + hdmidisplay and kbd. Chromium was launched via xinitrc and startx to open `http://moode.local`, a moode player somewhere else on the LAN. Chromium ver 130.0.6723.58.

Without `--disable-gpu` the moode WebUI showed some screen tearing and the circular sliders on the volume and time knobs were not visible. Adding `--disable-gpu` fixed those issues. Screen shots of the WebUI showing what the knobs look like are at https://moodeaudio.org

__/home/.xinitrc__
```
# Turn off display power management
xset -dpms
# Screensaver timeout in secs or 'off' for no timeout
xset s 600

# Capture screen size
SCREENSIZE="$(kmsprint | awk '$1 == "FB" {print $3}' | awk -F"x" '{print $1","$2}')"

# Launch chromium browser
chromium-browser --app="http://moode.local/" \
--window-size="$SCREENSIZE" \
--window-position="0,0" \
--enable-features="OverlayScrollbar" \
--no-first-run \
--disable-infobars \
--disable-session-crashed-bubble \
--kiosk
```

In your test config you would just need a moode player somewhere on your LAN that you can open in the Desktop Browser.

XECDesign, 2024-10-31:
Do you see rendering issues if you load the page from a default PiOS Desktop install? Does switching to labwc vs X11 make a difference (can be done via `raspi-config`? Or in that same environment you got now, but with only 'chromium-browser http://moode.local/'? Does having a WM like OpenBox running make any difference?

I can test those cases later as well, but narrowing it down a bit could help.

moodeaudio, 2024-10-31:
I've never used PiOS Desktop but I'll give it a try this morning and get back to you.

XECDesign, 2024-11-04:
Ah yeah, I can reproduce this with my previous setup as well. It just wasn't as obvious on other websites.

I can't see any clear reason why it might be doing this, but I can have a bit more of a dig later.

XECDesign, 2024-11-04:
Comparing the reports from chrome://gpu between working and non-working cases, it looks like ANGLE selects the wrong GL implementation. If you run with `--use-angle=gles`, you should retain GPU acceleration without the rendering artifacts.

XECDesign, 2024-11-04:
Same behaviour on stock Raspberry Pi OS image if it's switched to X11. So yeah, I think that's it - the issue is that ANGLE selects OpenGL rather than OpenGL ES when running under X11.

The problem also goes away if I launch chromium in the minimal environment under labwc.

XECDesign, 2024-11-04:
I had a quick look but gave up trying to figure out how Chromium figures out which ANGLE implementation it prefers. Using labwc or specifying `--use-angle=gles` would the suggested workaround for now.

I'll leave it open for now, but in general we are probably only going to support labwc going forward.

moodeaudio, 2024-11-05:
Tests this morning using a 3B+ and HDMI display confirm that `--use-angle=gles` works.

XECDesign, 2024-11-05:
I don't think suddenly dropping X11 support for something that should work is a good idea, so I'll leave it open until we've had a chance to take a look at it properly.

Thanks for spotting and helping narrow down the issue.

XECDesign, 2024-11-06:
I've added `--use-angle=gles` to the launch script and a `--no-gl-override` options for people who don't want that.

closed 2024-11-06 by XECDesign