RPi-Distro/chromium #51: Can't disable low memory warning in kiosk mode

entozoon, 2024-10-18:
Running chromium on say, a raspberry pi 3 model A in kiosk mode kicks out a warning dialog that you can't prevent.
 
The error message here in [/usr/bin/chromium-browser](https://github.com/RPi-Distro/chromium/blob/6b522e37f4bcb41fa8aafc5cf9e8ed913d324395/debian/scripts/chromium#L59) is kicked out later in that file [here](https://github.com/RPi-Distro/chromium/blob/6b522e37f4bcb41fa8aafc5cf9e8ed913d324395/debian/scripts/chromium#L59).

I'd recommend a flag that lets you prevent this warning because, well, not only does it actually run reasonably well (depending on your purposes), but it shouldn't have interstitial dialogs when booting into kiosk mode.

Often kiosk mode devices will just boot straight into a website for display purposes and intentionally don't have a keyboard/mouse to be able to skip this dialog.

[image]

popcornmix, 2024-10-18:
Reading the script shows there is a command line option `--no-memcheck` to disable the warning.

XECDesign, 2024-10-18:
Closing since the requested flag is already provided

closed 2024-10-18 by XECDesign

entozoon, 2024-10-18:
Nice one thanks