RPi-Distro/chromium #54: Chromium exists Kiosk mode after resolution change or monitor on/off

lentomajava, 2024-11-18:
**Description**

We have a problem with the new raspberry image with wayland/labwc when using the chromium browser in Kiosk mode.
When we turn off the monitor chromium drops out of full screen mode into a window and the kiosk mode is no longer full screen.
This is not the case with X11 installation.

Used chromium version 1:130.0.6723.116-1~deb12u1+rpt1

**Expected Behaviour**

Chromium should stay in full-screen Kiosk mode.

**Steps to reproduce the problem**

Here are the commands to reproduce the issue:
``` bash
/usr/bin/chromium-browser --enable-logging --v=1 --kiosk --enable-features=OverlayScrollbar --ozone-platform=wayland https://www.raspberrypi.com/
```
[image]

``` bash
# page displayed full screen, in kiosk mode
wlr-randr --output HDMI-A-1 --mode 1920x1080@60Hz --off
# page turns into a small window
wlr-randr --output HDMI-A-1 --mode 1920x1080@60Hz --on
```
[image]

``` bash
$ cat .config/chromium/chrome_debug.log
2486:2486:1118/114345.584357:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.579] Display: EVENT: wayland_screen.cc:198 Displays updated, count: 0
[2486:2486:1118/114345.688798:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.688] Display: EVENT: wayland_screen.cc:157 Displays updated, count: 1
[2486:2486:1118/114345.689214:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.688] Display: EVENT: wayland_screen.cc:160 Display[47] bounds=[0,0 1920x1080], workarea=[0,0 1920x1080], scale=1, rotation=0, panel_rotation=0 external detected
[2486:2486:1118/114345.690084:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.689] Display: EVENT: wayland_screen.cc:157 Displays updated, count: 1
[2486:2486:1118/114345.690356:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.690] Display: EVENT: wayland_screen.cc:160 Display[47] bounds=[0,0 1920x1080], workarea=[0,0 1920x1080], scale=1, rotation=0, panel_rotation=0 external detected
[2486:2486:1118/114345.690844:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.690] Display: EVENT: wayland_screen.cc:157 Displays updated, count: 1
[2486:2486:1118/114345.691037:VERBOSE1:device_event_log_impl.cc(203)] [11:43:45.690] Display: EVENT: wayland_screen.cc:160 Display[47] bounds=[0,0 1920x1080], workarea=[0,0 1920x1080], scale=1, rotation=0, panel_rotation=0 external detected

System Information
------------------

Raspberry Pi 4 Model B Rev 1.5
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ecda2579ebbc125b321984c571e3128f6fd959d4, stage4

Linux <name> 6.6.60-v8+ #1814 SMP PREEMPT Mon Nov 11 16:18:28 GMT 2024 aarch64 GNU/Linux
Revision        : b03115
Serial          : 1000000037b5249e
Model           : Raspberry Pi 4 Model B Rev 1.5
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Nov 11 2024 15:48:13
Copyright (c) 2012 Broadcom
version 903570ba72a9e117f92e5499de439f59dd96e417 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Package version information
---------------------------
[... 139 lines cut ...]
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
device_tree=-
overlay_prefix=overlays/
hdmi_cvt:0=
hdmi_cvt:1=
hdmi_edid_filename:0=
hdmi_edid_filename:1=
hdmi_timings:0=
hdmi_timings:1=

EEPROM
------

BOOTLOADER: up to date
   CURRENT: ma 21.10.2024 14.24.54 +0000 (1729520694)
    LATEST: ma 21.10.2024 14.24.54 +0000 (1729520694)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2711/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

XECDesign, 2024-11-18:
This seems to happen with geany as well, so it's looking like a labwc issue. 

@cillian64, also, the newer version of labwc in the internal repo doesn't seem to turn the monitor off via wlr-randr at all.

XECDesign, 2024-11-18:
https://github.com/labwc/labwc/issues/2362

cillian64, 2024-11-18:
> @cillian64, also, the newer version of labwc in the internal repo doesn't seem to turn the monitor off via wlr-randr at all.

Are you sure something else hasn't changed on your end? I just installed `libwlroots-0.18=0.18.1-1+rpt3` and `labwc=0.8.1-1+rpt1`, ran `wlr-randr --output HDMI-A-1 --off`, and my display turned off.

XECDesign, 2026-09-15:
Closing since I'm tidying up, labwc has moved on a fair bit since this was reported and it was never a chromium issue. If it's still relevant, please re-open [here](https://github.com/raspberrypi/trixie-feedback).

closed 2026-09-15 by XECDesign