RPi-Distro/chromium #57: White flash when playing mp4 videos

piotr-cz, 2025-06-17:
When playing mp4 videos, I have noticed that:

- there is a white flash just before video starts
- after few hours of playing different videos, chromium crashes with a page

  **Aw, Snap!**
  Something went wrong while displaying this web page.
  Error code: 5

  (Apparently this caused by [out-of memory issue](https://chromium.googlesource.com/chromium/src.git/+/refs/heads/main/chrome/browser/ui/sad_tab.cc#137))

This happens at least on chromium versions `137.0.7151.55` and `136.0.7103.92` on X11.

When starting chromium from command line (or with `--enable-logging` flag), at the time when video flashes, such log entries are produced:

```
[ERROR:gpu/command_buffer/service/shared_image/shared_image_factory.cc:906] Could not find SharedImageBackingFactory with params: usage: Gles2Read|RasterRead|DisplayRead, format: (Y_UV, 420, 8unorm, ExtSamplerOff), share_between_threads: 0, gmb_type: shared_memory, debug_label: MediaGmbVideoFramePoolMappableSI_Pid:0_Pid:0
[ERROR:gpu/ipc/service/shared_image_stub.cc:240] SharedImageStub: Unable to create shared image
[ERROR:gpu/command_buffer/service/shared_image/shared_image_manager.cc:401] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
```

Issue may be reproduced by visiting https://developer.mozilla.org/en-US/docs/Web/HTML/Reference/Elements/video page

On chromium `130.0.6723.116` there is no problem, and different log entry: 
```
[ERRPR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
```

I'll try to narrow down what's causing this and post updates here

might be because:
- https://github.com/RPi-Distro/chromium/commit/7a2a0cf0bc58a81d2b2cd19dd64d779693052981

BTW, starting chromium with `--no-gl-override` flag doesn't help

piotr-cz, 2025-06-18:
I've tried to open [MDN video](https://developer.mozilla.org/en-US/docs/Web/HTML/Reference/Elements/video) (right click, Open video in new tab -> `https://67b26f76-7fe0-4093-8d86-57b12ea0463a.mdnplay.dev/shared-assets/videos/flower.webm`)

The differences in DevTools Media panel are on Messages tab

On Raspberry PI with Chromium `137.0.7151.55-3`

- Cannot select *D3D11VideoDecoder* for video decoding
- Cannot select DecryptingVideoDecoder for video decoding
- Selected VpxVideoDecoder for video decoding...

On Windows with Google Chrome `137.0.7151.104`

- Cannot select *V4L2VideoDecoder* for video decoding
- Cannot select DecryptingVideoDecoder for video decoding
- Selected VpxVideoDecoder for video decoding...

XECDesign, 2025-06-19:
I don't think we can offer much support for X11 going forward. We've switched to Wayland a while ago and won't be investing further time into X11 in Trixie, so I think this is probably a wontfix.

@jc-kynesim @cillian64, would you happen to know what might be causing this?

cillian64, 2025-06-19:
I'm afraid I don't have any ideas.  I agree that if it's X11-specific (and also probably an issue in chromium given it didn't happen in older versions) I think it's a wontfix.

jc-kynesim, 2025-06-19:
Is this a Pi4 or a Pi5 or is it the same on both? If Pi4 only I'm going to guess that Chrome has got better at tyring to allocate appropriate V4L2 buffers but some part of the complete path is still broken and so falls back to a different decoder leaking something on the way. If both then dunno.